luke-jr/bfgminer #594: Mining sha-256 and scrypt with Dualminer

dtenney, 2015-01-14:
Sorry if this is in the documentation, but does bfgminer support mining both sha-256 and scrypt with a ASIC Dualminer DMU1-1C? And if so, are there any examples I can follow for setup.

BTW, Great product, keep up the good work!

Thanks!

David

nwoolls, 2015-01-14:
First, launch the SHA-2 miner instance:

```bfgminer -S noauto -S dualminer:all -o hostname -u username -p password```

Then launch the Scrypt miner instance with the --set-device argument:

```bfgminer -S noauto -S dualminer:all --set-device dualminer:dual_mode=1 --scrypt -o hostname -u username -p password```

Note that, as seen above, running in Dual Mode will result in a slower Scrypt hashrate: approximately 30Kh/s rather than 70Kh/s.

closed 2015-01-14 by nwoolls

dtenney, 2015-01-14:
Thanks for the information!

Any ideas why the Dualminer shows no processing for sha-256?
 DMU 0:       | 293.6/305.0/ 0.00Mh/s | A:0 R:0+0(none) HW:0/none

I've tried the dualminer by itself and with another standard sha-256 ASIC miner.
 BES 0:       | 255.3/295.5/446.4Mh/s | A:0 R:0+0(none) HW:0/none
 DMU 0:       | 331.3/323.8/ 0.00Mh/s | A:0 R:0+0(none) HW:0/none

The scrypt side works fine.
 DMU 0:       |  67.7/ 66.5/ 62.2kh/s | A:4 R:1+0( 17%) HW:0/none

With and without the scrypt process running, the sha-256 never processes.

Thanks!

nwoolls, 2015-01-14:
Well that is hashing SHA256 at only 300 Mh/s, so if you are hashing on a high diff coin / pool that may take a while to get an accepted share. How long have you waited?

Beyond that the driver was written for the original (revision A) sticks. I'm not sure if it's been tested with rev C sticks.

dtenney, 2015-01-14:
I've waited 36 hours. The other ASIC has been hashing the whole time.

I noticed that rev C does not have the toggle switch on the board like rev A, and whether or not the second bfgminer process is running, the scrypt process still shows the 70kh/s.

nwoolls, 2015-01-14:
Neither the BES or the DMU found a share though. See the A:0 next to both. The BES just uses lower diff shares for the 3rd hashrate while the DMU driver does not.

The problem is the same though - neither of your ASICs are finding shares on whatever SHA256 coin you are mining.

dtenney, 2015-01-14:
Ahh sorry, that wasn't a good example as I had just restarted the process.

That being said, rev C of the chip on the website is saying that the dualminer only processes scrypt, even though the dualminer comes up with two devices within the Linux environment.

http://www.dualminer.com/DualMiner-USB-2_p_18.html

Thanks for your help and sorry for wasting your time!

dtenney, 2015-01-14:
Interesting enough.. The only thing I can see that is different from the rev A and the rev C is the switch to change the voltage.

http://www.dualminer.com/DualMiner-USB_p_16.html