luke-jr/bfgminer #598: solo mining doesn't work with bitcoind 0.10.0-rc4

e477, 2015-02-07:
After upgrading to bitcoind 0.10.0-rc4, solo mining with bfgminer 5.0.0 no longer works:

    Testing pool http://localhost:8332
    Testing pool http://localhost:8332/#allblocks
    blktmpl error: Unrecognized block version, and not allowed to reduce or force it
    HTTP request failed: The requested URL returned error: 404 Not Found
    blktmpl error: Unrecognized block version, and not allowed to reduce or force it
    HTTP request failed: The requested URL returned error: 404 Not Found
    Closing socket for stratum pool 1
    Closing socket for stratum pool 0

luke-jr, 2015-02-08:
If you update libblkmaker from git, it has v3 block support. Will release 5.0.1 with it soon.

MarvinOl, 2015-02-09:
Will you be implementing this for Litecoin too? Litecoin is basically finishing steps of moving to corresponding version 0.10 of Bitcoin so that means no more getwork RPC (https://github.com/litecoin-project/litecoin/commits/exp-0.10-mark15).

luke-jr, 2015-02-09:
GBT is Bitcoin-specific.

luke-jr, 2015-02-19:
Fixed by 2f338cb

closed 2015-02-19 by luke-jr