luke-jr/bfgminer #606: libncurses-dev and ./config error

dowsettmark, 2015-03-11:
Hi,
-you are dealing with a bit of a noob here:
build on RPi-Raspbian

I ran into errors installing libncurses-dev last night and when trying to run ./configure - "command not recognized"

More than likely its me, but I noticed the latest commit on the config file was 15hrs ago (at time of attempt)?

luke-jr, 2015-03-11:
Just install it from aptitude... this isn't related to BFGMiner from what you've said.

closed 2015-03-11 by luke-jr

dowsettmark, 2015-03-11:
Ok, sorry for the time waste.

So run: sudo aptitude bfgminer

dowsettmark, 2015-03-11:
Sudo aptitude install bfgminer

luke-jr, 2015-03-11:
from README...

sudo aptitude install build-essential autoconf automake libtool pkg-config libcurl4-gnutls-dev libjansson-dev uthash-dev libncursesw5-dev libudev-dev libusb-1.0-0-dev libevent-dev libmicrohttpd-dev

Then follow the other instructions in README to compile.