luke-jr/bfgminer #608: autogen.sh fails cloning libblkmaker

wilson0x4d, 2015-03-12:
I have tried this on three different machines now (x86 and ARM7 architectures) and get the same result.

It appears the remote for 'libblkmaker' refers to gitorious using the git: protocol (rather than HTTPS) and it may be that gitorious is fialing to authorize anonymous users. If I manually clone from the command-line I can only clone if I use 'HTTPS'.

This appears to work around the issue, but I wind up having to do this every time I clone bfgminer for build. Would be nice if the origin configured in the repo could be corrected so it works for the widest audience.

Thanks. 

Here are my commands and the output:
```
root@buildsrv ~ $ rm -rf bfgminer
root@buildsrv ~ $ git clone https://github.com/luke-jr/bfgminer.git
Cloning into 'bfgminer'...
remote: Counting objects: 46656, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 46656 (delta 2), reused 0 (delta 0), pack-reused 46650
Receiving objects: 100% (46656/46656), 36.48 MiB | 712 KiB/s, done.
Resolving deltas: 100% (29723/29723), done.
root@buildsrv ~ $ cd bfgminer
root@buildsrv ~/bfgminer $ ./autogen.sh
Getting submodules...
Submodule 'ccan' (git://git.ozlabs.org/~ccan/ccan) registered for path 'ccan-upstream'
Submodule 'knc-asic' (git://github.com/KnCMiner/knc-asic) registered for path 'knc-asic'
Submodule 'libbase58' (git://github.com/luke-jr/libbase58.git) registered for path 'libbase58'
Submodule 'libblkmaker' (git://gitorious.org/bitcoin/libblkmaker.git) registered for path 'libblkmaker'
Cloning into 'ccan-upstream'...
remote: Counting objects: 16804, done.
remote: Compressing objects: 100% (5225/5225), done.
remote: Total 16804 (delta 12183), reused 15554 (delta 11341)
Receiving objects: 100% (16804/16804), 12.74 MiB | 1.16 MiB/s, done.
Resolving deltas: 100% (12183/12183), done.
Submodule path 'ccan-upstream': checked out '8b0bdb090e2882aa431e89f4bc7aa4736e9e2838'
Cloning into 'knc-asic'...
remote: Counting objects: 898, done.
remote: Total 898 (delta 0), reused 0 (delta 0), pack-reused 898
Receiving objects: 100% (898/898), 2.76 MiB | 1.53 MiB/s, done.
Resolving deltas: 100% (586/586), done.
Submodule path 'knc-asic': checked out 'e75c9ba362466546a83f1e705b62510a0e0afc6c'
Cloning into 'libbase58'...
remote: Counting objects: 167, done.
remote: Total 167 (delta 0), reused 0 (delta 0), pack-reused 167
Receiving objects: 100% (167/167), 29.15 KiB, done.
Resolving deltas: 100% (95/95), done.
Submodule path 'libbase58': checked out 'b6adca8ebb12962374a9cd1f6b543186f0cf6a40'
Cloning into 'libblkmaker'...
fatal: The remote end hung up unexpectedly
Clone of 'git://gitorious.org/bitcoin/libblkmaker.git' into submodule path 'libblkmaker' failed
```

And to highlight the problem/fix:
```
root@buildsrv ~/bfgminer $ git clone git://gitorious.org/bitcoin/libblkmaker.git
Cloning into 'libblkmaker'...
fatal: The remote end hung up unexpectedly
root@buildsrv ~/bfgminer $ git clone https://gitorious.org/bitcoin/libblkmaker.git
Cloning into 'libblkmaker'...
remote: Counting objects: 687, done.
remote: Compressing objects: 100% (683/683), done.
remote: Total 687 (delta 427), reused 0 (delta 0)
Receiving objects: 100% (687/687), 136.03 KiB | 119 KiB/s, done.
Resolving deltas: 100% (427/427), done.
root@buildsrv ~ $ git --version
git version 1.7.10.4
root@buildsrv ~ $ uname -a
Linux buildsrv 3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux
```

I'm waiting for the build to complete at this point, but it looks to be working.

luke-jr, 2015-03-12:
GitLab bought Gitorious and appears to be simply shutting it down.

wilson0x4d, 2015-03-13:
To be totally transparent, I've not pointed at Gitorious myself. This is the state of bfgminer on a fresh clone of master from from this repo on github.

Should the bfgminer source tree here on github be updated them to point its libblkmaker submodule path at github instead? (there appears to be https://github.com/bitcoin/libblkmaker but I'm not involved enough to know if this is a legit branch.)

daryltucker, 2015-03-23:
On many git hubs (GitHub/Bitbucket, etc), `https` is for READ-ONLY, while `git` access infers you have some explicit permission to read/write/administer the repository.

So, when using git as submodules/depends, one shoulduse `https` vs `git` for reading.

luke-jr, 2015-03-23:
Actually, the git protocol only supports reading... writing/push generally is done over https or ssh. Gitorious is closing, so the submodule now uses GitHub.

closed 2015-03-23 by luke-jr

daryltucker, 2015-03-23:
Hey, not to argue but I'm having issues cloning with the latest commit.

https://github.com/luke-jr/bfgminer/pull/611

luke-jr, 2015-03-23:
@daryltucker What issues?

daryltucker, 2015-03-23:
Any repository on Bitbucket or Github where the git:// (ssh) protocol is used.  Because the owners of the projects on those hosts haven't given me read/write/admin access to their repository, I am not allowed access.

Interesting point about the embedded devices, but I wish I could make it clear the issue I'm having because I think you'd understand.

When I connect to BB/GH, I use a specific key.  If I don't have read/write/admin access (explicitly through git) I am denied access.  The only way around this is for me to use https:// when pulling in a read-only fashion.

Last night two of the repos failed to clone.  It works for me now, but my intent is not to break it for everybody else.

luke-jr, 2015-03-23:
The git:// protocol is *not* ssh, and does not support keys, or even authentication. It is a strictly read-only protocol. http://git-scm.com/book/en/v2/Git-on-the-Server-The-Protocols#The-Git-Protocol

daryltucker, 2015-03-23:
Okay, I apologize for not being clear.

My issue was that ssh was being triggered on some of the repositories and I was denied access because of that.

I suppose it's the difference between:
* `git@github.com`
* `git://github.com`

I thought the `git://github.com` went over ssh, but you're telling me otherwise.

hehe

luke-jr, 2015-03-23:
Right. If you're trying to fetch via SSH, then you must have manually changed the remotes.

daryltucker, 2015-03-23:
I think it might have had more to do with `.gitmodules` being pulled into `.git/config` after initialization.

After a completely fresh clone of `bfgminer`, I see you're right that a fix is no longer needed.

daryltucker, 2015-03-27:
Fresh device, fresh clone:

    Cloning into 'libblkmaker'...
    fatal: Could not read from remote repository.

    Please make sure you have the correct access rights
    and the repository exists.
    Clone of 'git://gitorious.org/bitcoin/libblkmaker.git' into submodule path 'libblkmaker' failed

Steps to Reproduce:
1. `git clone https://github.com/luke-jr/bfgminer.git`
1. `cd bfgminer`
1. `git checkout bfgminer-5.1.0`
1. `./autogen.sh`

daryltucker, 2015-03-27:
I'm not trying to be a jerk, but I'm noticing the [docs use https:// vs git://](http://git-scm.com/book/en/v2/Git-Tools-Submodules).

Not to say that this is them saying it's the preferred method.  I just wonder what actual best-practice is, since I've always used https:// and you believe git:// should be used.

luke-jr, 2015-03-28:
Can't change old tags... you'll have to either clone libblkmaker manually, or use a branch.

daryltucker, 2015-03-28:
Oh, gotcha.

arijitdutta67, 2018-03-26:
Having the same issue with github repository of bfgminer . After doing git clone from git hub, the ./autogen.sh is not working. Here are the details:
~/bitcoin_code/bfgminer$ ./autogen.sh 
Getting submodules...
Cloning into 'ccan-upstream'...
fatal: unable to connect to git.ozlabs.org:
git.ozlabs.org[0: 103.22.144.67]: errno=Connection timed out
git.ozlabs.org[1: 2401:3900:2:1::2]: errno=Network is unreachable

Clone of 'git://git.ozlabs.org/~ccan/ccan' into submodule path 'ccan-upstream' failed