luke-jr/bfgminer #609: Failed to ioctl (1) in lowl-vcom.c windows_usb_get_root_hub_path():622: The parameter is incorrect

lashworth, 2015-03-16:
Failed to ioctl (1) in lowl-vcom.c windows_usb_get_root_hub_path():622: The parameter is incorrect

I'm using Win7 Pro 64bit. I have gridseed based 5.2MH miners.  The miners do operate, but the error message is annoying if nothing else...  What can I do to fix this?

luke-jr, 2015-03-16:
Identify why IOCTL_USB_GET_NODE_CONNECTION_NAME fails with ERROR_INVALID_PARAMETER

lashworth, 2015-03-16:
I'm running on Win 7 pro, 64 bit.

I'd be happy to dig deeper but I'm stumped, which does not happen easily. How would you suggest I proceed?

luke-jr, 2015-03-16:
It occurs during a scan of the USB bus, and indicates a USB hub scan cannot proceed/complete. If you get it repeatedly, --no-hotplug may help.

lashworth, 2015-03-16:
The error shows up just once during start up. And does not appear afterward.  Thank-you so much for your help!