luke-jr/bfgminer #610: Document how to start BFGminer with a specific device

ELLIOTTCABLE, 2015-03-21:
So, the README isn't very explanatory here. Reading,

>     --device|-d <arg>   Enable only devices matching pattern (default: all)

and,

>     --scan|-S <arg>     Configure how to scan for mining devices

and,

>     By default, BFGMiner will scan for autodetected devices. If you want to prevent
>     BFGMiner from doing this, you can use "-S noauto". If you want to probe all
>     serial ports, you can use "-S all"; note that this may write data to non-mining
>     devices which may then behave in unexpected ways!
>     
>     On Linux, <arg> is usually of the format /dev/ttyUSBn
>     On Mac OS X, <arg> is usually of the format /dev/cu.usb*
>     On Windows, <arg> is usually of the format \\.\COMn
>     (where n = the correct device number for the device)

I've tried each of the following:

    bfgminer -S none -d /dev/cu.usbserial-4321DEADBEEF
    bfgminer -S none -d 4321DEADBEEF
    bfgminer -S /dev/cu.usbserial-4321DEADBEEF
    bfgminer -S bfl@/dev/cu.usbserial-4321DEADBEEF

In the latter two cases, it started normally, connected to all seven devices, and started mining. No differences. In the first two, it connected to all seven devices, but each one was listed as ‘OFF.’ (Strangely enough, the temperatures all continued climbing. No idea why.)

nwoolls, 2015-03-21:
I think this should do it:

```bfgminer -S bfl:/dev/cu.usbserial-4321DEADBEEF -d bfl@/dev/cu.usbserial-4321DEADBEEF```

The ```-S``` argument tells BFGMiner which devices to scan/probe for miners and which types of devices to scan for. The ```-d``` argument tells BFGMiner which devices to use once it starts.

luke-jr, 2015-03-21:
Not sure where you got -S none from, or how the docs can be improved?

ELLIOTTCABLE, 2015-03-21:
@nwoolls hm. Okay, that works! Thanks! (Especially for `noauto`.)

@luke-jr Ah! I confused `-S noauto` for `-S none`. That's entirely on me!

And from the above, I suppose it should be pretty clear that the README's explanations could be improved for all of the following:

 - What arguments the `--device` / `-d` option takes, or what format those arguments should follow;
 - Why the `-d` option takes <tt>driver<em><strong>@</strong></em></tt> syntax, while the `-S` option actually takes the `<tt>driver<em><strong>:</strong></em></tt>` syntax (which the README contradicts!),
 - Why I need *both* of those options declared to restrict the connection to a single USB device? You'd think (well, I'd think, at least) think that either one or the other would preform that task.
 - That I need to specify *both* `-S noauto -S bfl:/dev/cu.usbserial-4321DEADBEEF` to connect to only one device.

In addition, given this warning,

> Q: Can I configure BFGMiner to mine with different login credentials or pools
for each separate device?
> A: No such feature has been implemented to support this.

… I think it'd be nice to add a short section to the README about running multiple BFGminers in parallel to circumvent this, noting some of what @nwoolls helped me figure out above. (=

luke-jr, 2015-03-21:
* Both accept the @path syntax, and :path is supported by -S only for backward compatibility.
* You don't need -d if you're using -S to restrict what devices are probed.

Would you be interested in updating the documentation to clarify these things? ;)

ELLIOTTCABLE, 2015-03-21:
Totally! I'm leaving for Hawai'i in a few hours, otherwise this Issue would have been a pull-request instead. :P

Re: `-S` without `-d`, pretty sure I tried that (as mentioned above), with no luck. Will try again when I get back; but it's Working Now, and I don't want to screw it up when I have to be on a plane soon! :D

alletro, 2016-01-13:
Can somebody tell me why  this is not working ?
I have tried more USB ports but not working....

bfgminer -S grd:/dev/ttyUSB1-48F372493133 -d grd@/dev/ttyUSB1-48F372493133

luke-jr, 2016-01-13:
There's no such device/driver using the letters "grd"...

alletro, 2016-01-13:
Sorry, something gets wrong wit past some text.
this is what I tried.

bfgminer -S bfl:/dev/ttyUSB1-48F372493133 -d bfl@/dev/ttyUSB1-48F372493133

I see only all my miners when I use bfgminer m+ option "all"
With option m+ "auto" I see  nothing.
Like to start them seperate.

luke-jr, 2016-01-13:
Those are very unusual device filenames. What do you get from:

    bfgminer -d? -D

alletro, 2016-01-13:
from the comment above:

nwoolls  commented on Mar 21 2015 

 bfgminer -S bfl:/dev/cu.usbserial-4321DEADBEEF -d bfl@/dev/cu.usbserial-4321DEADBEEF

luke-jr, 2016-01-13:
Those are Mac filenames. You need to see what they are on your system.

alletro, 2016-01-13:
Whats a good common Linux-syntax for the  Mac file above ?
I am just starting with Linux and ther are so maney different solutions in  how to files on the net.

luke-jr, 2016-01-13:
bfgminer -d?

This command should tell you what options are available...

alletro, 2016-01-14:
output: bfgminer -d?

Devices detected:
0 devices listed

must look why nothing is showing up while by option scan all
bfgminer is finding the miners and works.

maybe an good option  when option scan all or auto put de diagnostics as info in a file ?

regards and thx for helping

VanessaE, 2017-05-14:
Using bfgminer 5.4.2 from the Debian sid repository, here.

bfgminer does not properly interpret any standard pattern when specifying devices to scan. I want to restrict it from trying to scan the whole set of `/dev/ttyACM*` devices, as I also have a 3d printer which shows up under that pattern.  If bfgminer so much as breathes in its direction, the printer's host software throws an error and occasionally disconnects outright, so I want bfgminer to scan only what I tell it to.  Normally, I would use `-S gridseed:all` with no `-d` switch.  This of course scans everything, upsets the printer host, but otherwise finds and activates the gridseeds, and runs them normally.

I have tried every permutation of `-S` and `-d` with every form of pattern that I can come up with.  Here is everything that failed:

* `-S "gridseed:/dev/gridseed*"` (or with `@`, and/or `.*` or `+*`), with no `-d` parameter, ignores the printer, no devices found.
* `-S "gridseed:gridseed*"` with no `-d` parameter ignores the printer, no devices found.
* `-d "gridseed*"` with no `-S` switch ignores the printer, no devices found.
* `-d "/dev/gridseed*" ` with no `-S` switch ignores the printer, no devices found.
* `-S noauto -d "/dev/gridseed*" ` ignores the printer, no devices found.
* `-S gridseed:all -d "/dev/gridseed*"` scans everything, irritates the printer host, shows me all of the gridseeds at the top in the device list area, says it's "initializing" all of them, but at the same time it tells me at in the scrolling area at the bottom that it's disabling them all.
* `-S "/dev/gridseed*" -d "/dev/gridseed*"` ignores the printer, no devices found.  The same is true if I use single quotes, or no quotes and backslash-escaped asterisks.
* `-S "gridseed@/dev/gridseed*" -d "/dev/gridseed*"` ignores the printer, no devices found.
* `-S "gridseed@/dev/gridseed*" -d "gridseed@/dev/gridseed*"` ignores the printer, no devices found.
* `-S "gridseed:/dev/gridseed*" -d "gridseed@/dev/gridseed*"` ignores the printer, no devices found.
* `-S noauto -d gridseed@/dev/gridseed1` (note it's one specific device) ignores the printer, no devices found.

And finally, here's what was successful:

`-S "gridseed@/dev/gridseed1"` ignores the printer, finds /dev/gridseed1, and activates and runs it normally.

Building on that, I was able to activate all 12 of my gridseeds, without interfering with the printer, by giving multiple `-S` switches:

`-S "gridseed@/dev/gridseed1" -S "gridseed@/dev/gridseed2" -S "gridseed@/dev/gridseed3" ... `

That works.

kcw-grunt, 2017-12-22:
I spent the last few days sorting this out.  Here are my steps (Ubuntu+ FBM2 miners):
1. Write a 50-naming.rule in /etc/udev/rules.d that symlinks the device serial number to a unique name
2.Add  scan for the symlinked device(s) e.g. in the .start_moonlander.sh script:
-S /dev/FBM2###### --set-device /dev/FBM2######:clock=700
 3. This is working for me.
Although @luke-jr , I'm still puzzled about the -d=128 in the default script provided.  Is that to limit the number of devices? What is the cost/benefit if I go higher or lower? How can I match the dev symlink with the generic 'MLD X' at runtime?

jstefanop, 2017-12-22:
@kcw-grunt @luke-jr is the author of bfgminer, I am the author of Moonlander 2's driver integration. The d=128 flag is actually a pool specific flag thats set in the password field. This sets initial diff to scrypt 128. Since the moonlander 2 miner is "slow" compared to the large ASICs it will take forever to submit a share to the pool using its default diff, and newer users think the device is not working. 

This sets it to a lower diff initialy so shares are submitted more often.

jonaspirit, 2019-07-13:
@ please help to configure my moonlander2 coz i already connected to litecoin mining pool but my moonlander2 device is not communating or not detected.I dont know what to do.

thanks!!!!

jonaspirit, 2019-07-13:
@ i have a moonlander2 but it shows in the computer monitor no Divice found: press "M" and "+" to add, what should i do next? I`m using windows 7 computer.
Thank `s

jstefanop, 2019-07-13:
Windows 7 is probably part of the problem. If you dont see the devices under your device mangers (in the COM section) that means the silicon labs driver is not installed. They might not even support windows 7 at this point though.

jonaspirit, 2019-07-13:
Thank you for your kind reply, actually i`am not good in computer but i just try to have a miner to add a little incomecoz to tell you honestly i`am working in a goverment hospital as maintenance but,unfortunately I`am the lowest paid worker .Yup buy the way i try to set up moonlander 2 in my wife compute windows 10, 64 bit and i already seethe moonlander software in my device manager but still add device "M" and "+" to add .Hope you can give easy guide in order to finish this project.
thank you again

jonaspirit, 2019-07-29:
Good day sir! How to setup my moonlander2 in solo mining litecoin using windows7.thank you so mch

 
  On Sat., 13 Jul. 2019 at 10:12 a.m., jonas colina<colinaj12@yahoo.com> wrote:    Thank you for your kind reply, actually i`am not good in computer but i just try to have a miner to add a little incomecoz to tell you honestly i`am working in a goverment hospital as maintenance but,unfortunately I`am the lowest paid worker .Yup buy the way i try to set up moonlander 2 in my wife compute windows 10, 64 bit and i already seethe moonlander software in my device manager but still add device "M" and "+" to add .Hope you can give easy guide in order to finish this project.
thank you again

lubosz, 2021-02-02:
I'm getting
```
/dev/ttyACM0 is already in use by another process
```
Which is a Zigbee radio (smart home). Setting `-S noauto` does not help.

How do I make bfgminer ignore that device?