luke-jr/bfgminer #613: Gridseed Blade Degrade Cluster Degrades After An hour

computeronix, 2015-04-08:
I have a Gridseed blade cluster (4 gridseed blades x 2 USB each = total of 8 USB devices) plugged into a raspberry PI utilizing Minera 4.1.0 and the latest Build of  bfgminer 5.1.0-36-g8b5d36b.

The command I utilize is bfgminer --scrypt -S noauto -S gridseed:/dev/ttyACM0 --set gsd:clock=825 --queue 0 --scan-time 1 --expiry 1 --retries=1

This will allow one USB to run and it runs at about 2.4 MH/s or so and after about an hour the hashrate still reports this amount but the device flags as dead.  All 8 USBs do the exact same thing.

However, I have two other blade clusters and they are running solid using the exact same commands/software.

Any ideas?  I would chalk it up to hardware failure but 8 USB devices all at once?   Any recommended commands / debugging logs / etc to pull for research?

Thank you so much again!!!  Great product!!

computeronix, 2015-04-08:
One difference I noticed between the working cluster and the degraded cluster is the raspberry pi version is different

Working:   ARMv7 Processor rev 5 (v7l) (Raspberry PI 2)
Degraded: ARMv6-compatible processor rev 7 (v6l) (Raspberry PI Model B+)

Think the processor difference would matter?  The CPU/Memory/Disk usage is all low across the board.

luke-jr, 2015-04-08:
Raspberry Pi (1) does have a USB bus known to be broken/defective. I don't know if that was fixed in Pi2...

computeronix, 2015-04-08:
Interesting. This may make sense then as the other two miners are rasp 2

I ordered a rasp 2 and once I get it I will replace this exact miner with it and i will report back.

If the degradation is resolved then that must be the issue

Thank you

computeronix, 2015-04-18:
Switching to a Raspberry 2 resolved the issue