luke-jr/bfgminer #614: antminer u3 duplicating shares

piotrm0, 2015-04-22:
Running bfgminer with default options using antminer u3 results in frequent duplication as in the log below: 

[2015-04-22 05:44:08] Accepted 04fb2ade AMU 0b Diff 51/32
[2015-04-22 05:44:08] Rejected 04fb2ade AMU 0b Diff 51/32 (duplicate)
[2015-04-22 05:44:08] Rejected 04fb2ade AMU 0b Diff 51/32 (duplicate)
[2015-04-22 05:44:08] Rejected 04fb2ade AMU 0b Diff 51/32 (duplicate)

Running --benchmark results in many duplications as well.

Is this normal behaviour?

luke-jr, 2015-04-22:
From git?

piotrm0, 2015-04-22:
Yes, cloned yesterday.

Another thing I forgot to mention, the problem always appears in groups of 4, the first is accepted, the 3 that follow are duplicates. 4 is also the number of chips in this device which might or might not be a coincidence.

piotrm0, 2015-04-23:
Another bit of information: I'm getting this problem on Mac OS X, but not on raspberry pi with raspbian.

luke-jr, 2015-04-23:
Ah, yes. I would expect a problem like this on BSD/Mac... Someone will need to port stuff - the driver relies on epoll (Linux) or millisecond timeouts (Windows).

piotrm0, 2015-04-23:
Presumably this applies to driver-icarus as well as I see that is what driver-antminer relies on? I will look into adding use of kqueue if I have some free time.

luke-jr, 2015-04-23:
Well, the U3 is the only driver-icarus-based device that works so fast that the POSIX-standard 1 decisecond timeout is too slow.