luke-jr/bfgminer #615: erupters and antminers question

SalvorinFex, 2015-05-15:
Greetings

I have 3 Erupters and 3 antminers, 2 1.6 and one 2.0 I have been manually adding the erupters to the device list and I just ran across another thread to add the antminers, --scan-serial amu:all but that doesn't add the 2.0.

Is there a command line that I can add to the linux version for my Pi that will scan for all the erupters and antminers?

Thank you for a great program.

luke-jr, 2015-05-15:
AFAIK U2 behaves the same as U1 in this respect, but I have never had one to test with, so I can't really help.

SalvorinFex, 2015-05-15:
okay, well it is a U1.  All 3 are U1s, at least that's what the eBay posting said.

What about a command line addition for the erupters?

luke-jr, 2015-05-15:
```
-S erupter:all -S antminer:all
```
Should work

SalvorinFex, 2015-05-15:
When I did that it popped up with the main window and said: NO DEVICES FOUND

luke-jr, 2015-05-15:
Be sure you have the -S uppercase...

SalvorinFex, 2015-05-15:
Yeah, I copied and pasted.