luke-jr/bfgminer #618: Gridseed USB driver issue

SalvorinFex, 2015-05-24:
I have looked for the right USB drivers to work with the Gridseed 3355 but I can't seem to get one to work for bfgminer.  The miner is on COM4 and I have tried autoscan via the command line and adding it with the M+ but neither work.

Thank you

Sal

SalvorinFex, 2015-05-26:
I would like to get these miners up and running, Do you have any advice for me on this issue?

nwoolls, 2015-05-26:
What drivers did you install and on what OS? The GridSeed ASICs require a custom CDC driver on Windows. You can review a list of ASICs and their drivers (with downloads) here:

https://github.com/nwoolls/MultiMiner/wiki/Drivers

After the proper drivers are installed, the following command should list your ASICs:

```bfgminer -d? -S noauto -S gridseed:all --scrypt```

SalvorinFex, 2015-05-27:
I am running Windows 10 but just for testing, probably either ending up on Winodws 7 or Server 2008  Also, I wanted to mine BTC not Scrypt, even though I know that is not recommended.
What does the -d do for bfgminer?

nwoolls, 2015-05-27:
Mining SHA256 is not supported with the current GridSeed code in BFGMiner.

SalvorinFex, 2015-05-27:
OHH, that might be part of my problem then.

SalvorinFex, 2015-05-31:
This issue may be closed, Thanks.

closed 2015-05-31 by luke-jr