luke-jr/bfgminer #619: OSX 10.10: OpenCL needs VM Protection Flag?

ryanmatthews, 2015-06-12:
Hi,

I'm trying to run bfgminer on OS X 10.10 using OpenCL for scrypt mining. Things seem to detect okay but segfault when bfgminer tries to actually use the GPU. The kernel outputs the following message to syslog:

2015-06-11 6:05:39.000 PM kernel[0]: GPU Access disallowed due to VM protection attribute.

The crash report header (below) confirms this. I tried running as root but still no dice. It seems like to support OpenCL on OS X the software may need to "request" the correct VM privileges, but I'm not knowledgeable enough to dig that deep into this.

Process:               bfgminer [36890]
Path:                  /usr/local/bin/bfgminer
Identifier:            bfgminer
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [545]
Responsible:           iTerm [364]
User ID:               501

Date/Time:             2015-06-11 17:59:32.901 -0700
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        74A720DE-EE74-0C67-60E1-C53DC68BE8D6

Time Awake Since Boot: 1100000 seconds

Crashed Thread:        6  Dispatch queue: opencl_runtime

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018

luke-jr, 2015-06-12:
Sounds like something went wrong earlier. Can you get a backtrace, and/or at least a debug log?

ryanmatthews, 2015-06-12:
debug log below. I ran the following command:

bfgminer -S opencl:auto --debug --benchmark --debuglog 2>debug.log

debug.log:

 [2015-06-11 18:46:55] setrlimit: Changed soft fd limit from 2560 to 1024 (FD_SETSIZE=1024; hard limit=unlimited)
 [2015-06-11 18:46:55] Timers: Using gettimeofday
 [2015-06-11 18:46:55] Global quota greatest common denominator set to 1
 [2015-06-11 18:46:55] Generated target 00000000ffffffffffffffffffffffffffffffffffffffffffffffffffffffff
 [2015-06-11 18:46:55] Started bfgminer 5.2.0-unknown
 [2015-06-11 18:46:55] usb_devinfo_scan: Error getting USB string 2 (iManufacturer) from usb:253:001: LIBUSB_ERROR_OTHER
 [2015-06-11 18:46:55] usb_devinfo_scan: Error getting USB string 1 (iProduct) from usb:253:001: LIBUSB_ERROR_OTHER
 [2015-06-11 18:46:55] usb_devinfo_scan: Error getting USB string 2 (iManufacturer) from usb:250:001: LIBUSB_ERROR_OTHER
 [2015-06-11 18:46:55] usb_devinfo_scan: Error getting USB string 1 (iProduct) from usb:250:001: LIBUSB_ERROR_OTHER
 [2015-06-11 18:46:55] hidapi_try_lib: Successfully loaded libhidapi.dylib
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Apple Mikey HID Driver" serial "(null)"
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Ryans Mouse" serial "04-db-56-65-14-cc"
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Apple Keyboard" serial "(null)"
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Apple IR" serial "(null)"
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Microsoft LifeChat LX-6000" serial "(null)"
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Apple Keyboard" serial "(null)"
 [2015-06-11 18:46:55] hid_devinfo_scan: Found "Apple Wireless Trackpad" serial "d8-a2-5e-f7-1d-22"
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:001 (path=(null), vid=05ac, pid=8006, manuf=(null), prod=(null), serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:253:001 (path=(null), vid=05ac, pid=8006, manuf=(null), prod=(null), serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:002 (path=(null), vid=0424, pid=2514, manuf=(null), prod=(null), serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:003 (path=(null), vid=0a5c, pid=4500, manuf=Apple Inc., prod=BRCM2046 Hub, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:005 (path=(null), vid=05ac, pid=1006, manuf=Apple, Inc., prod=Keyboard Hub, serial=000000000000)
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:006 (path=(null), vid=045e, pid=077f, manuf=Microsoft, prod=Microsoft LifeChat LX-6000, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:007 (path=(null), vid=05ac, pid=8215, manuf=Apple Inc., prod=Bluetooth USB Host Controller, serial=78CA39C42E4A)
 [2015-06-11 18:46:55] lowlevel_scan: Found usb device at usb:250:010 (path=(null), vid=05ac, pid=0220, manuf=Apple, Inc, prod=Apple Keyboard, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid:Bluetooth_05ac_030e_5ef71d22 (path=Bluetooth_05ac_030e_5ef71d22, vid=05ac, pid=030e, manuf=Apple, prod=Apple Wireless Trackpad, serial=d8-a2-5e-f7-1d-22)
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid:USB_05ac_0220_fa132000 (path=USB_05ac_0220_fa132000, vid=05ac, pid=0220, manuf=Apple, Inc, prod=Apple Keyboard, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid:USB_045e_077f_fa140000 (path=USB_045e_077f_fa140000, vid=045e, pid=077f, manuf=Microsoft, prod=Microsoft LifeChat LX-6000, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid:USB_05ac_8242_fd120000 (path=USB_05ac_8242_fd120000, vid=05ac, pid=8242, manuf=Apple Computer, Inc., prod=Apple IR, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid:USB_05ac_0220_fa132000 (path=USB_05ac_0220_fa132000, vid=05ac, pid=0220, manuf=Apple, Inc, prod=Apple Keyboard, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid:Bluetooth_05ac_030d_566514cc (path=Bluetooth_05ac_030d_566514cc, vid=05ac, pid=030d, manuf=Apple, prod=Ryans Mouse, serial=04-db-56-65-14-cc)
 [2015-06-11 18:46:55] lowlevel_scan: Found hid device at hid: (path=, vid=0000, pid=0000, manuf=(null), prod=Apple Mikey HID Driver, serial=(null))
 [2015-06-11 18:46:55] lowlevel_scan: Found vcom device at dev_t:05000011 (path=/dev/cu.SerialAdapter-DevB, vid=0000, pid=0000, manuf=(null), prod=(null), serial=(null))
 [2015-06-11 18:46:55] CL Platform 0 vendor: Apple
 [2015-06-11 18:46:55] CL Platform 0 name: Apple
 [2015-06-11 18:46:55] CL Platform 0 version: OpenCL 1.2 (Feb 27 2015 01:29:10)
 [2015-06-11 18:46:55] Platform 0 devices: 1
 [2015-06-11 18:46:55] 	0	ATI Radeon HD 5670
 [2015-06-11 18:46:55] Unable to load ati adl library
 [2015-06-11 18:46:55] Init GPU thread 0 GPU 0 virtual GPU 0
 [2015-06-11 18:46:55] CL Platform vendor: Apple
 [2015-06-11 18:46:55] CL Platform name: Apple
 [2015-06-11 18:46:55] CL Platform version: OpenCL 1.2 (Feb 27 2015 01:29:10)
 [2015-06-11 18:46:55] List of devices:
 [2015-06-11 18:46:55] 	0	ATI Radeon HD 5670
 [2015-06-11 18:46:55] Selected 0: ATI Radeon HD 5670
 [2015-06-11 18:46:55] Preferred vector width reported 4
 [2015-06-11 18:46:55] Max work group size reported 256
 [2015-06-11 18:46:55] Max compute units reported 5
 [2015-06-11 18:46:55] Max mem alloc size is 134217728
 [2015-06-11 18:46:55] initCl() finished. Found ATI Radeon HD 5670
 [2015-06-11 18:46:55] Waking up thread 0
[... cut ...]
 [2015-06-11 18:46:55] Pushing work 1 from pool 0 to hash queue
 [2015-06-11 18:46:55] Selecting pool 0 for work
 [2015-06-11 18:46:55] Generated benchmark header 00000000000000000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000030000007fffffff1700ffff00000000
 [2015-06-11 18:46:55] Generated benchmark work
 [2015-06-11 18:46:55] Pushing work 2 from pool 0 to hash queue
 [2015-06-11 18:46:55] Error -54: Running goffset test kernel (clEnqueueNDRangeKernel)

ryanmatthews, 2015-06-12:
backtrace from same run:

VM Regions Near 0x18:
--> 
    __TEXT                 0000000104004000-00000001040da000 [  856K] r-x/rwx SM=COW  /usr/local/bin/bfgminer

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8aaeb5c2 swtch_pri + 10
1   libsystem_pthread.dylib       	0x00007fff91dde7df sched_yield + 11
2   bfgminer                      	0x0000000104007849 current_pool + 89 (miner.c:1254)
3   bfgminer                      	0x000000010401eb9b main + 20219 (miner.c:13705)
4   libdyld.dylib                 	0x00007fff8b03a5c9 start + 1

Thread 1:: org.libusb.device-hotplug
0   libsystem_kernel.dylib        	0x00007fff8aaeb4de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8aaea64f mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff936ffeb4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff936ff37b __CFRunLoopRun + 1371
4   com.apple.CoreFoundation      	0x00007fff936febd8 CFRunLoopRunSpecific + 296
5   com.apple.CoreFoundation      	0x00007fff937b6671 CFRunLoopRun + 97
6   libusb-1.0.0.dylib            	0x000000010415f487 darwin_event_thread_main + 591
7   libsystem_pthread.dylib       	0x00007fff91ddd268 _pthread_body + 131
8   libsystem_pthread.dylib       	0x00007fff91ddd1e5 _pthread_start + 176
9   libsystem_pthread.dylib       	0x00007fff91ddb41d thread_start + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8aaf1232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff8aa37a6a _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8aaf094a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff91ddb40d start_wqthread + 13

Thread 4 Crashed:: Dispatch queue: opencl_runtime
0   libsystem_platform.dylib      	0x00007fff8772f35a OSAtomicAdd32Barrier$VARIANT$mp + 2
1   com.apple.opencl              	0x00007fff8a2a84dc 0x7fff8a2a6000 + 9436
2   com.apple.opencl              	0x00007fff8a2c60ab 0x7fff8a2a6000 + 131243
3   com.apple.opencl              	0x00007fff8a2c9645 0x7fff8a2a6000 + 144965
4   libdispatch.dylib             	0x00007fff8aa34c13 _dispatch_client_callout + 8
5   libdispatch.dylib             	0x00007fff8aa38365 _dispatch_queue_drain + 1100
6   libdispatch.dylib             	0x00007fff8aa39ecc _dispatch_queue_invoke + 202
7   libdispatch.dylib             	0x00007fff8aa376b7 _dispatch_root_queue_drain + 463
8   libdispatch.dylib             	0x00007fff8aa45fe4 _dispatch_worker_thread3 + 91
9   libsystem_pthread.dylib       	0x00007fff91ddd637 _pthread_wqthread + 729
10  libsystem_pthread.dylib       	0x00007fff91ddb40d start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff8aaf094a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff91ddb40d start_wqthread + 13

Thread 6:
0   libsystem_kernel.dylib        	0x00007fff8aaf094a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff91ddb40d start_wqthread + 13

Thread 7:: miner_OCL0
0   libsystem_kernel.dylib        	0x00007fff8aaeb5c2 swtch_pri + 10
1   libsystem_pthread.dylib       	0x00007fff91dde7df sched_yield + 11
2   bfgminer                      	0x0000000104054059 _applog + 649 (.miner.h:811)
3   bfgminer                      	0x0000000104064612 opencl_test_goffset + 546 (ocl.c:1018)
4   bfgminer                      	0x0000000104062ee5 opencl_load_kernel + 1061 (ocl.c:1117)
5   bfgminer                      	0x000000010405cea4 opencl_scanhash_get_kernel + 164 (driver-opencl.c:1749)
6   bfgminer                      	0x000000010405e2d8 opencl_scanhash + 120 (driver-opencl.c:1766)
7   bfgminer                      	0x0000000104034e30 minerloop_scanhash + 256 (deviceapi.c:266)
8   bfgminer                      	0x0000000104036450 miner_thread + 352 (deviceapi.c:767)
9   libsystem_pthread.dylib       	0x00007fff91ddd268 _pthread_body + 131
10  libsystem_pthread.dylib       	0x00007fff91ddd1e5 _pthread_start + 176
11  libsystem_pthread.dylib       	0x00007fff91ddb41d thread_start + 13

Thread 8:: watchpool
0   libsystem_kernel.dylib        	0x00007fff8aaf048a __semwait_signal + 10
[... 231 lines cut ...]
ReadOnly portion of Libraries: Total=151.9M resident=80.7M(53%) swapped_out_or_unallocated=71.1M(47%)
Writable regions: Total=70.4M written=948K(1%) resident=5540K(8%) swapped_out=4K(0%) unallocated=65.0M(92%)
 
REGION TYPE                      VIRTUAL
===========                      =======
CG shared images                    144K
CoreServices                       1028K
Dispatch continuations             8192K
IOKit                              1364K
Kernel Alloc Once                     4K
MALLOC                             46.5M
MALLOC (admin)                       32K
OpenCL                                4K
STACK GUARD                        56.0M
Stack                              13.1M
VM_ALLOCATE                         416K
__DATA                             7168K
__GLSLBUILTINS                     2588K
__LINKEDIT                         71.4M
__TEXT                             80.4M
__UNICODE                           552K
mapped file                        3892K
shared memory                         4K
===========                      =======
TOTAL                             292.3M

luke-jr, 2015-06-12:
Looks like an OpenCL driver bug. clEnqueueNDRangeKernel is failing with CL_INVALID_WORK_GROUP_SIZE, which is nonsense since the local_work_size is 1 (which is always valid per the spec). Furthermore, it appears as if it still queues the kernel despite this, so after BFGMiner frees the failed kernel, the driver's thread tries to access it and crashes the program.

closed 2015-06-12 by luke-jr