luke-jr/bfgminer #620: Git BFGMiner hangs

proline777, 2015-07-01:
fyi I'm a newb to pi and raspbian, so please bare with me, but have been running cpu/cgminer/gridseed duals and am comfortable with it that way.  Would like to change from CPU to Pi
I'm trying to add bfgminer to my Raspberry Pi 2 running raspbian
When I try to 
git clone git://github.com/luke-jr/bfgminer.git
I receive Cloning into "bfgminer"...
then it just sits there.  Waited for hrs.
I'm not sure what I'm doing wrong.

Please help!

luke-jr, 2015-07-01:
Try cloning from git://git.savannah.nongnu.org/bfgminer.git instead? If neither work, maybe there's something wrong with your Pi's internet connection?

proline777, 2015-07-01:
I started over again, reformatted, reinstalled.  Everything progressed faster now that I have an internet connection (was a bad net cable).  When it comes to git bfgminer, I'm still receiving errors.

When I go into the windows style interface (startx) I have internet access and can surf around.

I tried your suggestion with the alt clone and I can report that somethings not right.

error:
fatal: unable to connect to savannah.nongnu.org:
savannah.nongnu.org[0: 140.196.70.711]: errno=Connection times out.

proline777, 2015-07-02:
Is it possible to install BFGMiner from a USB flash drive?

luke-jr, 2015-07-02:
Sure, grab the release txz file from http://luke.dashjr.org/programs/bitcoin/files/bfgminer/latest/

closed 2015-10-20 by luke-jr