luke-jr/bfgminer #621: BFGMiner 5.2.0 appears to hang at startup

chazzTSC, 2015-07-08:
Windows 7 x64, Antminer U2 single USB stick, mined properly on BFGMiner 5.0.0 and earlier. Noted that effective hash rate was falling to half of average hash rate (2.03GH/s average, 0.72GH/s effective) and thought that the block change might be the cause. Updated to BFGMiner 5.2.0; first attempt to run gave me a black screen with text 0000; closed window, second and later attempts gave me a black screen with BFGMiner 5.2.0 starting up. No activity lights show on Antminer -- usually under 5.0.0 there's about one flash of the green lights per second.

Start with -D -T --verbose gives me a screen full of code ending with

 [2015-07-08 09:51:42] QueryDosDevice returned insufficent buffer error; enlarging to 4000

 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:002:002 (path=(null), vid=80
87, pid=0020, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:001:002 (path=(null), vid=80
87, pid=0020, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:002:005 (path=(null), vid=1b
cf, pid=0c31, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:002:003 (path=(null), vid=13
07, pid=0330, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:002:004 (path=(null), vid=10
c4, pid=ea60, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:001:003 (path=(null), vid=04
c5, pid=201d, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:001:001 (path=(null), vid=80
86, pid=3b34, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found usb device at usb:002:001 (path=(null), vid=80
86, pid=3b3c, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found vcom device at com:16 (path=\\.\COM16, vid=000
0, pid=0000, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found vcom device at com:6 (path=\\.\COM6, vid=0000,
 pid=0000, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found vcom device at com:1 (path=\\.\COM1, vid=0000,
 pid=0000, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] lowlevel_scan: Found vcom device at com:5 (path=\\.\COM5, vid=0000,
 pid=0000, manuf=(null), prod=(null), serial=(null))
 [2015-07-08 09:51:42] AMU: cgpu_set_defaults: Matched with set default: antminer:clock=x0
981
 [2015-07-08 09:51:42] AMU: cgpu_set_defaults: Matched with set default: antminer:clock=x0
981
 [2015-07-08 09:51:42] AMU: cgpu_set_defaults: Matched with set default: antminer:clock=x0
981
 [2015-07-08 09:51:42] AMU: Applying rule antminer:clock=x0981: Unknown option: clock
 [2015-07-08 09:51:42] antminer: Attempting to open \\.\COM6
 [2015-07-08 09:51:42] AMU: Applying rule antminer:clock=x0981: Unknown option: clock
 [2015-07-08 09:51:42] AMU: Applying rule antminer:clock=x0981: Unknown option: clock
 [2015-07-08 09:51:42] AMU: cgpu_set_defaults: Matched with set default: antminer:clock=x0
981
 [2015-07-08 09:51:42] antminer: Failed to open \\.\COM6
 [2015-07-08 09:51:42] antminer: Attempting to open \\.\COM1
 [2015-07-08 09:51:42] antminer: Attempting to open \\.\COM16
 [2015-07-08 09:51:42] AMU: Applying rule antminer:clock=x0981: Unknown option: clock
 [2015-07-08 09:51:42] antminer: Attempting to open \\.\COM5
 [2015-07-08 09:51:43] antminer: Test succeeded at \\.\COM16: got 000187a2
 [2015-07-08 09:51:43] \\.\COM16: Init: mode=long read_timeout_ms=75 limit=0ms Hs=4.761000
e-010
 [2015-07-08 09:51:43] antminer: Work division not specified - autodetecting
 [2015-07-08 09:51:43] antminer: Work division autodetection got 04c0fdb4 (=1)
 [2015-07-08 09:51:43] Found AMU 0 at \\.\COM16
 [2015-07-08 09:51:43] AMU 0: Init: baud=115200 work_division=0 fpga_count=0

luke-jr, 2015-07-08:
Please try to use http://luke.dashjr.org/tmp/code/webisect/webisect.php to identify the source of this regression. Pick any reasonably unique session name, and then on the next page:

Working commit: bfgminer-5.0.0

Broken commit: bfgminer-5.2.0

Continue to follow the instructions until you get a result, then paste the result here.

chazzTSC, 2015-07-08:
I understand the purpose, which is to refine the delta that's causing 
the issue, but as I don't recognize the effect of the deltas being 
reported, I'd end up simply skipping all of them. Which would not be 
useful to you, unfortunately.

Is there any other way to tackle this?

luke-jr, 2015-07-08:
I don't understand the problem you are having bisecting. Your problem is clearly either present or not... either it hangs, or it starts. Right?

chazzTSC, 2015-07-08:
Oh -- I can pull a binary commit and check it? I haven't used bisector 
before, didn't know that was what I was meant to do. It is very sparse 
on detail.

luke-jr, 2015-07-08:
webisect provides Windows binaries.

chazzTSC, 2015-07-08:
Sorry for being dense.

d50810f29f9a91e3303d512e8b16ef3945c51b4f is the first bad commit
commit d50810f29f9a91e3303d512e8b16ef3945c51b4f
Author: Luke Dashjr
Date: Thu Feb 26 09:07:01 2015 +0000

icarus: Replace decisecond-precision read_count with read_timeout_ms 
(millisecond precision) to handle faster devices like the Antminer U3 
that complete works in under 1ds

:100644 100644 7575642899d830643cc4e1501ce9f551e707338b 
2466fffdc896fbbcf834545fa7088a17eb470f04 M driver-antminer.c
:100644 100644 cf0d204e21412dcf141099ff1d16931110bb05e8 
5c81a397e619339e16c72ce243d0a0ad17d60c06 M driver-cairnsmore.c
:100644 100644 90df94dc10fbce615610e8fe0776298592d55cea 
5600260971b094b9437b0b7c5126793094189a53 M driver-dualminer.c
:100644 100644 dcf715476bc3ea6461dec2eed9543b6d6268683a 
4435d662abad17515c29b873b262bde821b06220 M driver-icarus.c
:100644 100644 7b1bf695cedf54d6b5d2f1ea74fa1693e3ffa67b 
937eebaa82bc1748dd523f63355439af21b183ea M driver-icarus.h
:100644 100644 09be0f9d9ebca3af2c9363bada4e10d6de30017e 
47b33dc50431f4dd6fb77916eb2969672aa12247 M driver-zeusminer.

luke-jr, 2015-07-09:
Can you get me the same -D -T --verbose using http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=tmp249 ?

chazzTSC, 2015-07-09:
Certainly.

End of the page:

  pid=0000, manuf=(null), prod=(null), serial=(null))
  [2015-07-08 21:11:19] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 21:11:19] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 21:11:19] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 21:11:19] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 21:11:19] antminer: Attempting to open \\.\COM1
  [2015-07-08 21:11:19] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 21:11:19] antminer: Attempting to open \\.\COM5
  [2015-07-08 21:11:19] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 21:11:19] antminer: Attempting to open \\.\COM16
  [2015-07-08 21:11:19] \\.\COM1 open: fd=3
  [2015-07-08 21:11:19] \\.\COM1 begin write1
  [2015-07-08 21:11:19] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 21:11:19] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 21:11:19] antminer: Attempting to open \\.\COM6
  [2015-07-08 21:11:19] \\.\COM5 open: fd=4
  [2015-07-08 21:11:19] \\.\COM5 begin write1
  [2015-07-08 21:11:19] \\.\COM5 end write1
  [2015-07-08 21:11:19] \\.\COM6 open: fd=-1
  [2015-07-08 21:11:19] antminer: Failed to open \\.\COM6
  [2015-07-08 21:11:19] \\.\COM5 begin read1
  [2015-07-08 21:11:19] \\.\COM1 end write1
  [2015-07-08 21:11:19] \\.\COM1 begin read1
  [2015-07-08 21:11:19] \\.\COM16 open: fd=5
  [2015-07-08 21:11:19] \\.\COM16 begin write1
  [2015-07-08 21:11:19] \\.\COM16 end write1
  [2015-07-08 21:11:19] \\.\COM16 begin read1
  [2015-07-08 21:11:19] \\.\COM16 end read1
  [2015-07-08 21:11:19] \\.\COM16 begin excess1
  [2015-07-08 21:11:19] \\.\COM1: No data in 0.100 seconds
  [2015-07-08 21:11:19] \\.\COM1 end read1
  [2015-07-08 21:11:19] \\.\COM1 begin excess1
  [2015-07-08 21:11:19] \\.\COM16 end excess1
  [2015-07-08 21:11:19] \\.\COM16 begin close1
  [2015-07-08 21:11:19] \\.\COM16 end close1
  [2015-07-08 21:11:19] antminer: Test succeeded at \\.\COM16: got 000187a2
  [2015-07-08 21:11:19] \\.\COM16: Init: mode=long read_timeout_ms=75 
limit=0ms Hs=4.761000
e-010
  [2015-07-08 21:11:19] antminer: Work division not specified - 
autodetecting
  [2015-07-08 21:11:19] \\.\COM1 end excess1
  [2015-07-08 21:11:19] \\.\COM1 begin close1
  [2015-07-08 21:11:19] \\.\COM1 end close1
  [2015-07-08 21:11:19] antminer: Test failed at \\.\COM1: get 00000000, 
should: 000187a2
  [2015-07-08 21:11:19] antminer: Work division autodetection got 
04c0fdb4 (=1)
  [2015-07-08 21:11:19] Found AMU 0 at \\.\COM16
  [2015-07-08 21:11:19] AMU 0: Init: baud=115200 work_division=0 
fpga_count=0

And there she hangs.

I can get more of the head of the log if you need it...

luke-jr, 2015-07-09:
Same thing again, please. http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=tmp249 is updated with a new build.

chazzTSC, 2015-07-09:
It does look a little different, I think... but it still freezes. If you 
need more of the back log let me know.

  [2015-07-08 23:04:24] lowlevel_scan: Found vcom device at com:16 
(path=\\.\COM16, vid=000
0, pid=0000, manuf=(null), prod=(null), serial=(null))
  [2015-07-08 23:04:24] lowlevel_scan: Found vcom device at com:6 
(path=\\.\COM6, vid=0000,
  pid=0000, manuf=(null), prod=(null), serial=(null))
  [2015-07-08 23:04:24] lowlevel_scan: Found vcom device at com:1 
(path=\\.\COM1, vid=0000,
  pid=0000, manuf=(null), prod=(null), serial=(null))
  [2015-07-08 23:04:24] lowlevel_scan: Found vcom device at com:5 
(path=\\.\COM5, vid=0000,
  pid=0000, manuf=(null), prod=(null), serial=(null))
  [2015-07-08 23:04:24] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 23:04:24] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 23:04:24] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 23:04:24] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 23:04:24] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 23:04:24] antminer: Attempting to open \\.\COM16
  [2015-07-08 23:04:24] AMU: cgpu_set_defaults: Matched with set 
default: antminer:clock=x0
981
  [2015-07-08 23:04:24] antminer: Attempting to open \\.\COM1
  [2015-07-08 23:04:24] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 23:04:24] AMU: Applying rule antminer:clock=x0981: Unknown 
option: clock
  [2015-07-08 23:04:24] antminer: Attempting to open \\.\COM6
  [2015-07-08 23:04:24] \\.\COM1 open: fd=3
  [2015-07-08 23:04:24] antminer: Attempting to open \\.\COM5
  [2015-07-08 23:04:24] \\.\COM6 open: fd=-1
  [2015-07-08 23:04:24] \\.\COM1 begin write1
  [2015-07-08 23:04:24] \\.\COM5 open: fd=4
  [2015-07-08 23:04:24] \\.\COM5 begin write1
  [2015-07-08 23:04:24] \\.\COM5 end write1
  [2015-07-08 23:04:24] antminer: Failed to open \\.\COM6
  [2015-07-08 23:04:24] \\.\COM5 begin read1
  [2015-07-08 23:04:24] \\.\COM1 end write1
  [2015-07-08 23:04:24] \\.\COM1 begin read1
  [2015-07-08 23:04:24] \\.\COM16 open: fd=5
  [2015-07-08 23:04:24] \\.\COM16 begin write1
  [2015-07-08 23:04:24] \\.\COM16 end write1
  [2015-07-08 23:04:24] \\.\COM16 begin read1
  [2015-07-08 23:04:24] \\.\COM16 end read1
  [2015-07-08 23:04:24] \\.\COM16 begin excess1
  [2015-07-08 23:04:24] \\.\COM16 end excess1
  [2015-07-08 23:04:24] \\.\COM16 begin close1
  [2015-07-08 23:04:24] \\.\COM16 end close1
  [2015-07-08 23:04:24] antminer: Test succeeded at \\.\COM16: got 000187a2
  [2015-07-08 23:04:24] \\.\COM16: Init: mode=long read_timeout_ms=75 
limit=0ms Hs=4.761000
e-010
  [2015-07-08 23:04:24] antminer: Work division not specified - 
autodetecting
  [2015-07-08 23:04:24] antminer: Work division autodetection got 
04c0fdb4 (=1)
  [2015-07-08 23:04:24] Found AMU 0 at \\.\COM16
  [2015-07-08 23:04:24] AMU 0: Init: baud=115200 work_division=0 
fpga_count=0

luke-jr, 2015-07-09:
That isn't from the new build... Check the filename begins with 9e255373f4cea8462ec132b2557798838bac3583

chazzTSC, 2015-07-09:
Sorry.

9e25537 seems to work. Should I just use that or did you want to push a 
finished version?

luke-jr, 2015-07-09:
I didn't make any functional changes to 9e25537, just logging... Please retry until it hangs :|

chazzTSC, 2015-07-09:
Oh - it's intermittent. I know, I should wash my mouth out with soap 
after saying the I word.

So: try it again until it hangs. Like this?

  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=4, buf, 1) => 0
  [2015-07-09 00:07:10] fd=4 timer_passed(1231461.734614, 1231461.734614)
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=4, 6)
  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=4, buf, 1) => 0
  [2015-07-09 00:07:10] fd=4 timer_passed(1231461.734614, 1231461.734614)
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=4, 5)
  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=4, buf, 1) => 0
  [2015-07-09 00:07:10] fd=4 timer_passed(1231461.734614, 1231461.734614)
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=4, 3)
  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=4, buf, 1) => 0
  [2015-07-09 00:07:10] fd=4 timer_passed(1231461.734614, 1231461.734614)
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=4, 2)
  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=4, buf, 1) => 0
  [2015-07-09 00:07:10] fd=4 timer_passed(1231461.734614, 1231461.734614)
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=4, 1)
  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=4, buf, 1) => 0
  [2015-07-09 00:07:10] fd=4 timer_passed(1231461.734614, 1231461.734614)
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=4, 0)
  [2015-07-09 00:07:10] read(fd=4, buf, 1) ...
  [2015-07-09 00:07:10] read(fd=3, buf, 1) => 0
  [2015-07-09 00:07:10] fd=3 timer_passed(1231461.739313, 1231461.739313)
  [2015-07-09 00:07:10] \\.\COM1: No data in 0.102 seconds
  [2015-07-09 00:07:10] \\.\COM1 end read1
  [2015-07-09 00:07:10] \\.\COM1 begin excess1
  [2015-07-09 00:07:10] \\.\COM16 end excess1
  [2015-07-09 00:07:10] \\.\COM16 begin close1
  [2015-07-09 00:07:10] \\.\COM16 end close1
  [2015-07-09 00:07:10] antminer: Test succeeded at \\.\COM16: got 000187a2
  [2015-07-09 00:07:10] \\.\COM16: Init: mode=long read_timeout_ms=75 
limit=0ms Hs=4.761000
e-010
  [2015-07-09 00:07:10] antminer: Work division not specified - 
autodetecting
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=5, 75)
  [2015-07-09 00:07:10] read(fd=5, buf, 1) ...
  [2015-07-09 00:07:10] \\.\COM1 end excess1
  [2015-07-09 00:07:10] \\.\COM1 begin close1
  [2015-07-09 00:07:10] \\.\COM1 end close1
  [2015-07-09 00:07:10] antminer: Test failed at \\.\COM1: get 00000000, 
should: 000187a2
  [2015-07-09 00:07:10] read(fd=5, buf, 1) => 1
  [2015-07-09 00:07:10] vcom_set_timeout_ms(fd=5, 27)
  [2015-07-09 00:07:10] read(fd=5, buf, 4) ...
  [2015-07-09 00:07:10] read(fd=5, buf, 4) => 4
  [2015-07-09 00:07:10] antminer: Work division autodetection got 
04c0fdb4 (=1)
  [2015-07-09 00:07:10] Found AMU 0 at \\.\COM16
  [2015-07-09 00:07:10] AMU 0: Init: baud=115200 work_division=0 
fpga_count=0

luke-jr, 2015-07-09:
Yeah, except .. the whole thing please :)

chazzTSC, 2015-07-09:
Sorry. Had to increase the screen buffer.

Better?

C:\Program Files\BFGMiner>bfgminer.exe --no-submit-stale --set-device 
antminer:clock=x0981
  -o http://mint.bitminter.com:8332 -u xxx -p AnythingGoesHere -S 
amu:all -D -T
  --verbose

  [2015-07-09 01:04:26] setrlimit: Not supported by platform
  [2015-07-09 01:04:26] Started bfgminer 5.2.0-2-g9e25537
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:002:001: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:001:001: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:001:003: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:002:004: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:002:003: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:002:005: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:001:002: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] usb_devinfo_scan: Error opening device 
usb:002:002: LIBUSB_ERROR_NO
T_SUPPORTED
  [2015-07-09 01:04:26] hidapi_try_lib: Couldn't load libhidapi.dll: unknown
  [2015-07-09 01:04:26] hidapi_try_lib: Loaded libhidapi-0.dll, but no 
devices enumerated;
trying other libraries
  [2015-07-09 01:04:26] hidapi_try_lib: Couldn't load 
libhidapi-hidraw.dll: unknown
  [2015-07-09 01:04:26] hidapi_try_lib: Couldn't load 
libhidapi-libusb.dll: unknown
  [2015-07-09 01:04:26] hid_devinfo_scan: Failed to load any hidapi library
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c mswin_devinfo
_scan():47: No more data is available.
  [2015-07-09 01:04:26] Failed to SetupDiEnumDeviceInterfaces in 
[... 510 lines cut ...]
  [2015-07-09 01:04:27] \\.\COM16 end excess1
  [2015-07-09 01:04:27] \\.\COM16 begin close1
  [2015-07-09 01:04:27] \\.\COM16 end close1
  [2015-07-09 01:04:27] antminer: Test succeeded at \\.\COM16: got 000187a2
  [2015-07-09 01:04:27] \\.\COM16: Init: mode=long read_timeout_ms=75 
limit=0ms Hs=4.761000
e-010
  [2015-07-09 01:04:27] \\.\COM1 end excess1
  [2015-07-09 01:04:27] \\.\COM1 begin close1
  [2015-07-09 01:04:27] \\.\COM1 end close1
  [2015-07-09 01:04:27] antminer: Test failed at \\.\COM1: get 00000000, 
should: 000187a2
  [2015-07-09 01:04:27] antminer: Work division not specified - 
autodetecting
  [2015-07-09 01:04:27] vcom_set_timeout_ms(fd=3, 75)
  [2015-07-09 01:04:27] read(fd=3, buf, 1) ...
  [2015-07-09 01:04:27] read(fd=3, buf, 1) => 1
  [2015-07-09 01:04:27] vcom_set_timeout_ms(fd=3, 27)
  [2015-07-09 01:04:27] read(fd=3, buf, 4) ...
  [2015-07-09 01:04:27] read(fd=3, buf, 4) => 4
  [2015-07-09 01:04:27] antminer: Work division autodetection got 
04c0fdb4 (=1)
  [2015-07-09 01:04:27] Found AMU 0 at \\.\COM16
  [2015-07-09 01:04:27] AMU 0: Init: baud=115200 work_division=0 
fpga_count=0

luke-jr, 2015-07-09:
Ok, same link again with a new build. This time, it's a candidate fix. Please confirm the problem is consistently resolved, add -d? (that's a literal question-mark) to your options, and paste the full output from that so I can confirm the change is functioning as expected. Thanks for your patience!

chazzTSC, 2015-07-09:
No worries about the patience. I'm a developer myself, I know what a 
pain it is to not have a local test bed that replicates the problem.

83a7c5... goes as far as detecting the devices and then simply exits. Is 
that what the d? option does?

c:\Program Files\BFGMiner>bfgminer.exe --no-submit-stale --set-device 
antminer:c
lock=x0981 -o http://mint.bitminter.com:8332 -u xxxx -p AnythingGoesHere 
-S amu:all -D -T --verbose -d?

  [2015-07-09 08:18:28] setrlimit: Not supported by platform
  [2015-07-09 08:18:28] Started bfgminer 5.2.0-4-g83a7c53
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:002:001: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:001:001: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:001:003: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:002:004: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:002:003: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:002:005: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:001:002: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] usb_devinfo_scan: Error opening device 
usb:002:002: LIBUS
B_ERROR_NOT_SUPPORTED
  [2015-07-09 08:18:28] hidapi_try_lib: Couldn't load libhidapi.dll: unknown
  [2015-07-09 08:18:28] hidapi_try_lib: Loaded libhidapi-0.dll, but no 
devices en
umerated; trying other libraries
  [2015-07-09 08:18:28] hidapi_try_lib: Couldn't load 
libhidapi-hidraw.dll: unkno
wn
  [2015-07-09 08:18:28] hidapi_try_lib: Couldn't load 
libhidapi-libusb.dll: unkno
wn
  [2015-07-09 08:18:28] hid_devinfo_scan: Failed to load any hidapi library
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
lowl-mswin.c msw
in_devinfo_scan():47: No more data is available.
  [2015-07-09 08:18:28] Failed to SetupDiEnumDeviceInterfaces in 
[... 543 lines cut ...]
s=4.761000e-010
  [2015-07-09 08:18:28] antminer: Work division not specified - 
autodetecting
  [2015-07-09 08:18:28] vcom_set_timeout_ms(fd=3, 75)
  [2015-07-09 08:18:28] read(fd=3, buf, 1) ...
  [2015-07-09 08:18:28] \\.\COM1 end excess1
  [2015-07-09 08:18:28] \\.\COM1 begin close1
  [2015-07-09 08:18:28] \\.\COM1 end close1
  [2015-07-09 08:18:28] antminer: Test failed at \\.\COM1: get 00000000, 
should:
000187a2
  [2015-07-09 08:18:29] read(fd=3, buf, 1) => 1
  [2015-07-09 08:18:29] vcom_set_timeout_ms(fd=3, 23)
  [2015-07-09 08:18:29] read(fd=3, buf, 4) ...
  [2015-07-09 08:18:29] read(fd=3, buf, 4) => 4
  [2015-07-09 08:18:29] antminer: Work division autodetection got 
04c0fdb4 (=1)
  [2015-07-09 08:18:29] Found AMU 0 at \\.\COM16
  [2015-07-09 08:18:29] AMU 0: Init: baud=115200 work_division=0 
fpga_count=0
  [2015-07-09 08:18:29] Devices detected:
  [2015-07-09 08:18:29]  Device (driver=antminer; procs=1; path=\\.\COM16)
1 devices listed

c:\Program Files\BFGMiner>

luke-jr, 2015-07-09:
Yes, -d? lists devices detected. Running it how you normally do should work okay now?

chazzTSC, 2015-07-09:
Seems to. Of course, it did work once before the fix <wry>... but so far 
so good.

luke-jr, 2015-07-09:
If http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=bfgminer works, then please close this issue. :)

chazzTSC, 2015-07-10:
I will note that I got the line "0000" as the first log line, before 
"Probing for an alive pool".

This one reports its version number as 5.2.0-2-ga488b3f. It also seemsto 
be doing something odd with its average speeds: the Antminer is rated 
2.0 Gh/S, but the speed report says 3.74 / 3.14 / 2.27 Gh/s, which is 
somewhat impossible...

It is mining though. Did you want to look at the speed calculations 
before I closed the bug?

luke-jr, 2015-07-10:
No, speeds for these things is inherently guesswork/unreliable. Any other bugs would be unrelated to this bug, so need another issue anyway.

chazzTSC, 2015-07-10:
Okay. I will comment that 5.0.0 seemed to correctly report average and 
moving average raw Gh/s at 2.0 give or take 1%, while 5.2.0-2 seems to 
report it just about exactly twice that. It seems to have settled at 
3.97/ 3.91/ 1.97 Gh/s after 30 minutes.

closed 2015-07-10 by chazzTSC

luke-jr, 2015-07-10:
5.0.0 used a fixed timing based on knowledge of the U1/U2 expected speeds. Since U3 is vastly different, 5.2.0 instead tries to estimate.