luke-jr/bfgminer #626: BFGMiner 5.2.0-2 crashes at startup when pool not present

chazzTSC, 2015-07-17:
BFGMiner 5.2.0-2 crashes in Windows without showing an error message when a pool host can't be found. Running normally, the only line printed is the title line 

 [2015-07-16 17:48:09] Started bfgminer 5.2.0-2-ga488b3f

Running with -D -T --verbose, the final lines are

 [2015-07-16 17:42:54] initiate_stratum with sockbuf=0000000000000000
 [2015-07-16 17:42:54] Stratum connect failed with TLS to pool 0: Could not resolve host: us3.bitminter.com
 [2015-07-16 17:42:54] Stratum connect failed to pool 0: Could not resolve host: us3.bitminter.com
 [2015-07-16 17:42:54] Initiate stratum failed

Immediately following this line we get the Windows "This program has failed" pop-up.

Expected: Graceful failure with error message.

luke-jr, 2015-07-17:
I have no information to go on here. Does your BFGMiner directory have backtrace.dll? Do you get this problem with an official release (which 5.2.0-2-ga488b3f is not)?

chazzTSC, 2015-07-17:
The official release 5.2.0 cannot be tested because it wedges with my Antminer 2. (That bug is only a week old.) I can retry with 5.0.0, I think; the wedge is a regression, 5.0.0. shouldn't have it.

This distro does have backtrace.dll.

luke-jr, 2015-07-17:
Sure. If 5.0.0 doesn't have this problem, please create a session in http://luke.dashjr.org/tmp/code/webisect/webisect.php and give it:

* Working: bfgminer-5.0.0
* Broken: a488b3f

It will give you a series of custom builds to test, and finally a result. Post the result here.

If at any point you get a more detailed error, please also post it here.

chazzTSC, 2015-07-17:
Windows error message is useless:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	bfgminer.exe
  Application Version:	0.0.0.0
  Application Timestamp:	0a73f0ba
  Fault Module Name:	msvcrt.dll
  Fault Module Version:	7.0.7601.17744
  Fault Module Timestamp:	4eeb033f
  Exception Code:	c0000005
  Exception Offset:	0000000000001444
  OS Version:	6.1.7601.2.1.0.256.4
  Locale ID:	4105
  Additional Information 1:	64a3
  Additional Information 2:	64a3851c9a971b58dd0b3578e4f27516
  Additional Information 3:	57d1
  Additional Information 4:	57d1e37f41974d683c94c62b57cfde54

It appears that the issue was put in immediately after 5.0.0 - unless this is again an intermittent and I only got it working in 5.0.0 by luck.

b97f51db9617784b5998d41de7889e92a8026043 is the first bad commit
commit b97f51db9617784b5998d41de7889e92a8026043
Author: Luke Dashjr 
Date:   Sun Nov 30 01:48:21 2014 +0000

    configure: Avoid BASHisms for better portability

:100644 100644 09951a4604efdcfae398345c549dcbe2e0841e88 6fdc8434af5f9c54f726d9b9b0b86d172819601c M	configure.ac

luke-jr, 2015-07-17:
Yeah, I don't see any possibility that's the problem. :(

chazzTSC, 2015-07-17:
And yet 5.0.0. works. Just retested. Bitminter started looking for 512, and is asking for 64 at the moment; when 5.2.0-2 was working, it was asking for 16 initially, falling to 4. Clutching at straws: could it be to do with requested difficulty?.

chazzTSC, 2015-07-17:
Thought: Last cycle I saw a huge number of hardware errors (38%). This miner may be toast. Could that be the issue?

luke-jr, 2015-07-17:
Doubt it. If you run with -S noauto you should eliminate the device's relevance. Maybe see if the crash occurs with that, and if so, re-do the bisect (use a new session name)?

luke-jr, 2015-07-17:
Oh, it's also possible the bisect result could indicate a problem with a library... See if http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=bfgminer-4.0.0 behaves any differently from the official 4.0.0 binaries?

chazzTSC, 2015-07-17:
I don't think I have the official 4.0.0 binaries. Would you mean the 5.0.0? And does that change the webisect command line?

luke-jr, 2015-07-17:
Yes, sorry. Please change both numbers to 5.0.0 in that comment.

chazzTSC, 2015-07-17:
No, with great dismay I must report that they both work the same -- to be precise, both 5.0.0 and the webisect build actually do work. (They also both leave a stray "ocktemplate)" on screen in the status area until it scrolls out, but that's trivial.)