luke-jr/bfgminer #627: autogen.sh error on Raspberry PI 2

ilma1, 2015-07-23:
Hi,

When installing I get this:
Configure.ac :35: warning: macro  AM_PATH_LIBGCRYPT not found in library
Configure.ac :29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available

And there is not make file.

Best regards
Magnus

luke-jr, 2015-07-23:
Read README.

closed 2015-07-23 by luke-jr

WangleX, 2015-08-13:
1.Try : 
"apt-get remove autogen automake autoconf libtool m4"
2. Type ''whereis xxx"(xxx is one of autogen automake autoconf libtool m4)
to find out is there are these file still, then delete them all !
3.Type
"apt-get install autogen automake autoconf libtool m4"
to reinstall these packages
I've got the same errors as yours, then fixed them via this method. You can try it out! 
My platform is raspian on raspberry pi

monkins1010, 2018-03-09:
whats the fix for this if you have mingw?  and trying to compile bfgminer. 
how do i install libgcrypt20 on mingw?  thanks kindly