luke-jr/bfgminer #629: BFGMiner 5.2.0 - Antminer U3 - Intermittent USB connection issue

anto, 2015-08-20:
I am experiencing an intermittent USB connection issue on Antminer U3 with BFGMiner 5.2.0. The issue could be on CP2102 driver but I am not sure about that.

After running fine for about 13 hours, bfgminer lost the serial connection to Antminer U3. This happens randomly. The last time I remember, it was running for about 16 hours before that happened.

After I stopped bfgminer and restart it, it cannot detect /dev/ttyUSB0. I had to power down Antminer U3, power it back up and reconnect the USB cable for bfgminer to recognise the USB connection.

I don't see any messages which can help me figure out what is going on. So I really appreciate any help on this.

The dmesg only shows the following:
```
root@hp8530w:~# dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.18-1v0-hp8530w (anto@hp8530w) (gcc version 4.9.2 (Debian 4.9.2-10) ) #2 SMP Sat Jul 11 21:19:29 CEST 2015
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-3.18.18-1v0-hp8530w root=UUID=c0e1e620-5636-48f2-90dd-4d246d58b815 ro quiet
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
.
.
.
[1213606.695392] wlan0: associated
[1213606.695783] cfg80211: Calling CRDA for country: AT
[1213606.699571] cfg80211: Regulatory domain changed to country: AT
[1213606.699576] cfg80211:  DFS Master region: unset
[1213606.699578] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[1213606.699581] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[1213606.699584] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[1213606.699587] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2000 mBm), (0 s)
[1213606.699590] cfg80211:   (5490000 KHz - 5710000 KHz @ 80000 KHz), (N/A, 2700 mBm), (0 s)
[1213606.699592] cfg80211:   (57240000 KHz - 65880000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[1246538.915471] cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[1246538.920403] usb usb6-port1: disabled by hub (EMI?), re-enabling...
[1246538.920419] usb 6-1: USB disconnect, device number 41
[1246538.920444] cp210x ttyUSB0: usb_serial_generic_submit_read_urb - usb_submit_urb failed: -19
[1246538.921783] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[1246538.921813] cp210x 6-1:1.0: device disconnected
[1246539.160327] usb 6-1: new full-speed USB device number 42 using uhci_hcd
[1246539.319726] usb 6-1: New USB device found, idVendor=10c4, idProduct=ea60
[1246539.319737] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1246539.319743] usb 6-1: Product: CP2102 USB to UART Bridge Controller
[1246539.319748] usb 6-1: Manufacturer: Silicon Labs
[1246539.319754] usb 6-1: SerialNumber: 0001
[1246539.324756] cp210x 6-1:1.0: cp210x converter detected
[1246539.324988] usb 6-1: cp210x converter now attached to ttyUSB0
[1256149.580304] cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[1256149.645291] cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[1256149.664391] usb usb6-port1: disabled by hub (EMI?), re-enabling...
[1256149.664407] usb 6-1: USB disconnect, device number 42
[1256149.666682] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[1256149.666729] cp210x 6-1:1.0: device disconnected
[1256149.908310] usb 6-1: new full-speed USB device number 43 using uhci_hcd
[1256150.071344] usb 6-1: New USB device found, idVendor=10c4, idProduct=ea60
[1256150.071355] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1256150.071362] usb 6-1: Product: CP2102 USB to UART Bridge Controller
[1256150.071367] usb 6-1: Manufacturer: Silicon Labs
[1256150.071372] usb 6-1: SerialNumber: 0001
[1256150.076342] cp210x 6-1:1.0: cp210x converter detected
[1256150.076498] usb 6-1: cp210x converter now attached to ttyUSB0
[1267999.600424] usb 6-1: USB disconnect, device number 43
[1267999.600868] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[1267999.600907] cp210x 6-1:1.0: device disconnected
[1268018.852305] usb 6-1: new full-speed USB device number 44 using uhci_hcd
[1268019.014590] usb 6-1: New USB device found, idVendor=10c4, idProduct=ea60
[1268019.014601] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1268019.014607] usb 6-1: Product: CP2102 USB to UART Bridge Controller
[1268019.014613] usb 6-1: Manufacturer: Silicon Labs
[1268019.014618] usb 6-1: SerialNumber: 0001
[1268019.020516] cp210x 6-1:1.0: cp210x converter detected
[1268019.020897] usb 6-1: cp210x converter now attached to ttyUSB0
[... 41 lines cut ...]
Aug 20 09:58:31 hp8530w mtp-probe: bus: 6, device: 43 was not an MTP device
Aug 20 13:16:01 hp8530w kernel: [1267999.600424] usb 6-1: USB disconnect, device number 43
Aug 20 13:16:01 hp8530w kernel: [1267999.600868] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
Aug 20 13:16:01 hp8530w kernel: [1267999.600907] cp210x 6-1:1.0: device disconnected
Aug 20 13:16:20 hp8530w kernel: [1268018.852305] usb 6-1: new full-speed USB device number 44 using uhci_hcd
Aug 20 13:16:20 hp8530w kernel: [1268019.014590] usb 6-1: New USB device found, idVendor=10c4, idProduct=ea60
Aug 20 13:16:20 hp8530w kernel: [1268019.014601] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 20 13:16:20 hp8530w kernel: [1268019.014607] usb 6-1: Product: CP2102 USB to UART Bridge Controller
Aug 20 13:16:20 hp8530w kernel: [1268019.014613] usb 6-1: Manufacturer: Silicon Labs
Aug 20 13:16:20 hp8530w kernel: [1268019.014618] usb 6-1: SerialNumber: 0001
Aug 20 13:16:20 hp8530w kernel: [1268019.020516] cp210x 6-1:1.0: cp210x converter detected
Aug 20 13:16:20 hp8530w kernel: [1268019.020897] usb 6-1: cp210x converter now attached to ttyUSB0
Aug 20 13:16:20 hp8530w mtp-probe: checking bus 6, device 44: "/sys/devices/pci0000:00/0000:00:1d.2/usb6/6-1"
Aug 20 13:16:20 hp8530w mtp-probe: bus: 6, device: 44 was not an MTP device
root@hp8530w:~# 
```
As we can see on /var/log/messages above, there are 2 events of USB disconnections at 07:18:20 and 09:58:31 CEST. BFGminer kept running without serial connection to Antminer U3. I restarted bfgminer at 13:17:42 after I power cycled Antminer U3 as we can see on below picture.

[image]

The graph on "My Eligius" shows the following.

[image]

Assuming that the time stamp on the graph is on UTC, the time when I restarted bfgminer at 13:17:42 CEST looks aligned with the time stamp on the graph, i.e. after 11:00 UTC. But the time when bfgminer stopped getting the data from Antminer U3 at around 09:15 UTC or 11:15 CEST, does not seem to match any events on /var/log/messages.

luke-jr, 2015-08-20:
The U3 is known to crash/hang often - is it possible the USB intermittancy issues are coincidental and unrelated?

anto, 2015-08-20:
Yes. It could be coincidental and unrelated. But I am not really sure about that Luke.

I didn't take the debug log as it would create a huge log file as the problem usually happens after more than 10 hours. But if we looked at the screenshot of console below, Antminer U3 was still able to send the result from AMU 0b at 10:41:36 CEST (or 08:41:36 UTC) and it was still being accepted by your eligius mining pool as shown the graph above.

[image]

For whatever reason, it just stopped mining about 30 minutes after that. The LED mining indicator (the one close to the fan connector) was not illuminating any more as if there was no bfgminer process but the process was still running. The fact that restarting bfgminer causes it not to see Antminer U3 on /dev/ttyUSB0 and power cycling Antminer U3 solves the connection problem, I am suspecting there is an issue either on CP2102 driver or bfgminer itself. Perhaps you could implement a mechanism to detect the connection and declare the miner to be dead. I am not sure though how that could be done as I am not a programmer. At the moment, the only way to know that Antminer U3 is not mining any more is to look at the LED mining indicator or check "My User Stats Page" on "My Eligius".

luke-jr, 2015-08-20:
If you don't mind the downtime, can you see how it behaves with --benchmark?

anto, 2015-08-20:
No. I don't mind. For how long should I run that to get reasonable statistic?

luke-jr, 2015-08-20:
However long it takes to reproduce your issue, I guess. Debug log would be helpful.

anto, 2015-08-20:
Hello Luke,

I decided to run bfgminer with --benchmark for just a little bit more than 1 hour. The parameters that I am using (voltage=x775, clock=x0882 and timing=0.0162) came from my trials and errors in the last few days. I get more stable result at around 50 GH/s effective hash rate on that lower voltage and clock. I am not sure what you are after but please have a look on screenshot and debug log below. I hope you find what you are looking for. If not, please let me know more detail information to collect the log that you need, e.g. for how long, with what parameters, etc, as precise as possible.

Cheers,

Anto

[image]

https://minifora.eu/public/bfgminer/bfgminer_5.2.0_benchmark_debug_20Aug2015_2050.log.gz

anto, 2015-08-20:
Ooppss... I just sent my reply at almost the same time as you.

I can run the benchmark as long as we need in order to capture the problem. But if I would run the benchmark for more than 10 hours, the debug log file will be quite huge. Do you have any suggestion?

luke-jr, 2015-08-20:
Compression? :)

anto, 2015-08-20:
I was expecting better answer than that :)

For instance, is there any command to limit the debug output only to the messages related to the serial communication between BFGminer and Antminer U3 via USB interface? Do you provide API for debugging so that sniff the USB interface for taking specific control messages?

I will start the benchmarking to run as long as possible before I go to sleep tonight.

anto, 2015-08-20:
I changed my mind :)

I actually have 2 Antminer U3. I got one of them new from an online shop in Germany last week. The other one is a used one but the same batch 2. I just got it yesterday from an online flee market here in Austria. With the exact same parameters, the performance of both miners are quite identical.

So instead of using one of them to get the benchmark log, I installed BFGMiner 5.2.0 on my TP-Link TL-WDR3600 WiFi router which is running on OpenWRT. I connected the new miner (the one from the online shop) to my WiFi router and the used one to my PC. Both of them are running now and connected to eligius mining pool.

The USB issue that I reported here happened on my PC, either using the new or used miner. So if the USB connection of the one that is connected to my WiFi router will also experience the same problem, then we can rule out anything else but the BFGMiner 5.2.0 itself. Let's see about this tomorrow night.

luke-jr, 2015-08-21:
> For instance, is there any command to limit the debug output only to the messages related to the serial communication between BFGminer and Antminer U3 via USB interface?

No.

> Do you provide API for debugging so that we can sniff the USB interface for taking specific control messages?

--device-protocol-dump although it relies on drivers supporting it at present. Failing that, Wireshark should work on Linux.

anto, 2015-08-27:
After a few days of fiddling around, I came to the conclusion that there seems to be nothing we can do on BFGMiner side. For what ever reason, the Antminer U3 just suddenly hangs after hashing for more than 12 hours, in my case. The only solution that I know of is to power cycle it.

What I have done so far is to prevent both miners from running continuously using a cron job. I plug both miners into my WiFi router and run 2 separate BFGMiner processes so that I can do interleaving of stopping each BFGminer process, reset its USB port and start the process again. I initially stopped that for 15 minutes every 7 hours and 45 minutes. By doing that, I hope the ASIC chips can be cooled down for a while and anything on USB interface get flushed.

Both miners ran for more than a day and then one of them hung again and generating high pitch of beeping sounds in the morning. Following the discussion on bitcointalk.org, I disassembled both miners, cleaned and applied the best thermal compound that I could find. The miners looked more stable for almost  a day even after I increased the hash rate to 60 GH/s all-time average each. This morning, I reduced the "cooling and flushing" process to 5 minutes every 11 hours and 55 minutes.

Next step maybe to build a better cooling system and use automatic power switch controlled by a little Arduino board. But I think that is too much of hassles for mining purpose, so I am looking for more stable miner.

anto, 2015-08-28:
As I previously mentioned, this issue is more related to the instability problem on Antminer U3  itself. Therefore I am closing this issue to make the issue list cleaner.

closed 2015-08-28 by anto