luke-jr/bfgminer #632: Zeusminer not detected (debian jessie 64bit)

0xicl33n, 2015-09-13:
Ive done multiple compilations of bfgminer on multiple devices using a lot of command line options.

I have a zeusminer x3 thunder which works fine with the official cgminer from zeusminer (its 3.1.1)

I want to use bfgminer but it just will not detect it at all. Ive tried modifying the udev rules and using -S zeusminer:all 

The miner is detected in dmesg as an ftdi device and works perfectly fine when used with  the cgminer from the company. The ancient version of cgminer has no interface so it makes it hard to manage, or use at all for multipool.

[root@anvil[/b/bfgminer:bfgminer*]$./bfgminer --version
bfgminer 5.2.0-9-g484db74
  Lowlevel: ftdi hid pci pci:uio pci:vfio spi usb vcom
  Drivers: antminer avalon avalonmm bfx bifury bigpic bitforce bitforce:pci bitfury cairnsmore cointerra drillbit dualminer erupter gridseed hashbuster hashbusterusb hashfast icarus klondike littlefury modminer nanofury proxy proxy:getwork proxy:stratum rockminer twinfury x6500 zeusminer ztex
  Algorithms: SHA256d scrypt
  Options: ncursesw5

Linux anvil 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1+deb8u3 (2015-08-04) x86_64 GNU/Linux

[    2.504035] usb 3-3: new full-speed USB device number 2 using ohci-pci
[    2.722072] usb 3-3: New USB device found, idVendor=0403, idProduct=6001
[    2.722075] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.722078] usb 3-3: Product: FT232R USB UART
[    2.722080] usb 3-3: Manufacturer: FTDI
[    2.722082] usb 3-3: SerialNumber: A402XN6B
[    3.032031] usb 3-4: new full-speed USB device number 3 using ohci-pci
[    3.241072] usb 3-4: New USB device found, idVendor=058f, idProduct=9360
[    3.241075] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.241078] usb 3-4: Product: USB Reader
[    3.241080] usb 3-4: Manufacturer:
[    3.241082] usb 3-4: SerialNumber: 2004888

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 058f:9360 Alcor Micro Corp. 8-in-1 Media Card Reader
**Bus 003 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC**
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 413c:2105 Dell Computer Corp. Model L100 Keyboard
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

The board is detected as /dev/ttyUSB0 and seems to work just fine with the other program.. i just dont understand why bfgminer refuses to see it.

My options for launching bfgminer:
bfgminer --scrypt  -o stratum+tcp:/pool:info -u user -p pass -S zeusminer:/dev/ttyUSB0 --set zeusminer:clock=328 --set zeusminer:chips=128

I even tried

bfgminer --scrypt -S --benchmark -d? -D -S /dev/ttyUSB0 and get the following output:

--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 [2015-09-12 19:14:46] lowlevel_scan: Found pci device at pci:0000:00:13.1 (path=0000:00:13.1, vid=1002, pid=4375, manuf=(null), prod=(null), serial=(null))
 [2015-09-12 19:14:46] lowlevel_scan: Found pci device at pci:0000:00:13.0 (path=0000:00:13.0, vid=1002, pid=4374, manuf=(null), prod=(null), serial=(null))
 [2015-09-12 19:14:46] lowlevel_scan: Found pci device at pci:0000:00:12.0 (path=0000:00:12.0, vid=1002, pid=4379, manuf=(null), prod=(null), serial=(null))
 [2015-09-12 19:14:46] lowlevel_scan: Found pci device at pci:0000:00:01.0 (path=0000:00:01.0, vid=1002, pid=5a3f, manuf=(null), prod=(null), serial=(null))
 [2015-09-12 19:14:46] lowlevel_scan: Found pci device at pci:0000:00:00.0 (path=0000:00:00.0, vid=1002, pid=5950, manuf=(null), prod=(null), serial=(null))
 [2015-09-12 19:14:46] lowlevel_scan: Found vcom device at dev_t:00bc000000000000 (path=/dev/ttyUSB0, vid=0000, pid=0000, manuf=FTDI, prod=FT232R USB UART, serial=A402XN6B)
 [2015-09-12 19:14:46] zeusminer: Attempting to open /dev/ttyUSB0
 [2015-09-12 19:14:46] Acquired exclusive advisory lock on /dev/ttyUSB0
 [2015-09-12 19:14:47] /dev/ttyUSB0: No data in 0.501 seconds
 [2015-09-12 19:14:47] zeusminer: Test failed at /dev/ttyUSB0: get 00000000, should: 00038d26
 [2015-09-12 19:14:47] zeusminer: Attempting to open /dev/ttyUSB0
 [2015-09-12 19:14:47] Acquired exclusive advisory lock on /dev/ttyUSB0
 [2015-09-12 19:14:47] /dev/ttyUSB0: No data in 0.501 seconds
 [2015-09-12 19:14:47] zeusminer: Test failed at /dev/ttyUSB0: get 00000000, should: 00038d26
 [2015-09-12 19:14:47] Device rescan requested
 [2015-09-12 19:14:48] hid_devinfo_scan: Found "Dell USB Keyboard" serial "(null)"
 [2015-09-12 19:14:48] lowlevel_scan: Found usb device at usb:002:001 (path=(null), vid=1d6b, pid=0001, manuf=Linux 3.16.0-4-amd64 ohci_hcd, prod=OHCI PCI host controller, seria
l=0000:00:13.0)
 [2015-09-12 19:14:48] lowlevel_scan: Found usb device at usb:002:002 (path=(null), vid=413c, pid=2105, manuf=Dell, prod=Dell USB Keyboard, serial=(null))
 [2015-09-12 19:14:48] lowlevel_scan: Found usb device at usb:003:001 (path=(null), vid=1d6b, pid=0001, manuf=Linux 3.16.0-4-amd64 ohci_hcd, prod=OHCI PCI host controller, seria
l=0000:00:13.1)
 [2015-09-12 19:14:48] lowlevel_scan: Found usb device at usb:003:002 (path=(null), vid=0403, pid=6001, manuf=FTDI, prod=FT232R USB UART, serial=A402XN6B)
 [2015-09-12 19:14:48] lowlevel_scan: Found usb device at usb:003:003 (path=(null), vid=058f, pid=9360, manuf= , prod=USB Reader, serial=2004888)
[... 22 lines cut ...]
 [2015-09-12 19:14:48] lowlevel_scan: Found vcom device at dev_t:00bc000000000000 (path=/dev/ttyUSB0, vid=0000, pid=0000, manuf=FTDI, prod=FT232R USB UART, serial=A402XN6B)
 [2015-09-12 19:14:48] zeusminer: Attempting to open /dev/ttyUSB0
 [2015-09-12 19:14:48] Acquired exclusive advisory lock on /dev/ttyUSB0
 [2015-09-12 19:14:48] /dev/ttyUSB0: No data in 0.501 seconds
 [2015-09-12 19:14:48] zeusminer: Test failed at /dev/ttyUSB0: get 00000000, should: 00038d26
 [2015-09-12 19:14:48] zeusminer: Attempting to open /dev/ttyUSB0
 [2015-09-12 19:14:48] Acquired exclusive advisory lock on /dev/ttyUSB0

The output from the other cgminer looks like this when i use it instead (im not dumb enough to try and run both miner programs at once):

(5s):11.79M (avg):27.78Mh/s | A:962  R:3  HW:55  U:102.7/m  WU:27874.6/m
(5s):4.329M (avg):27.49Mh/s | A:971  R:3  HW:55  U:102.6/m  WU:27823.5/m
(5s):66.74M (avg):28.28Mh/s | A:981  R:3  HW:56  U:102.5/m  WU:27827.0/m
(5s):41.77M (avg):28.27Mh/s | A:985  R:3  HW:56  U:101.9/m  WU:27671.5/m
(5s):15.33M (avg):27.98Mh/s | A:995  R:3  HW:57  U:101.9/m  WU:27676.5/m
(5s):34.98M (avg):28.16Mh/s | A:1000  R:3  HW:57  U:101.3/m  WU:27499.8/m
(5s):12.84M (avg):27.88Mh/s | A:1011  R:3  HW:59  U:101.4/m  WU:27557.7/m
(5s):4.714M (avg):27.61Mh/s | A:1018  R:3  HW:59  U:101.1/m  WU:27462.0/m
(5s):54.75M (avg):28.16Mh/s | A:1034  R:3  HW:60  U:101.7/m  WU:27619.9/m
(5s):20.10M (avg):27.88Mh/s | A:1047  R:3  HW:61  U:102.0/m  WU:27699.9/m

and is in fact mining away happily

What is wrong with bfgminer?
 I also tried the one via aptitude and it didnt work either

Wolfwood5351, 2015-09-22:
This seems to be a common theme with FTDI devices. I'm running a Butterfly Labs device with the FT232H in it and I'm having the same issue.

0xicl33n, 2015-09-23:
i changed to https://github.com/dmaxl/cgminer/ with no problems. running just fine

kr1z1s, 2015-12-26:
There are ASIC company Ruminer. Within the chip A2, Zeus motherboard. There is a converter, UART-USB FTDI.
I use openwrt 15.05 and 2 different models of routers NEXX wt3020 and TpLINK 3020.
I compile openwrt with kernel modul FTDI and Bfg. Asic determined normal, port ttyUSB0. All ok.
BFG without parametr starts normally.
If i starts BFG with "-S /dev/ttyUSB0" BFG aborts and in /dev dir disappears /dev/ttyUSB0
Mystery  :(

With other computer device (Raspberry, Banana) BFG work fine, using the same FTDI modul and ZeusMiner drivers.