luke-jr/bfgminer #633: BFGMiner 5.3.0 sends result after new block was detected causing it to be rejected

anto, 2015-09-13:
As far as I understood, when a new block is being detected, BFGMiner should dump the current results and asks for work for the new block. It does not seem to be the case, because I saw the following messages:
```
[2015-09-13 16:02:47] Stratum from pool 0 detected new block
[2015-09-13 16:02:47] Pool 0 is hiding block contents from us
[2015-09-13 16:02:47] Rejected 000d52d1 CBM 0d pool 0 Diff 4.92k/256 (unknown-work)
```

luke-jr, 2015-09-14:
--no-submit-shares if you want that behaviour (although this message can still occur since the speed of networking is not instant).

luke-jr, 2015-09-14:
(Also note that behaviour has no benefits, and may some times have negatives...)

closed 2015-09-14 by luke-jr