luke-jr/bfgminer #640: BFGMiner 5.3.0 commit c000fc9 - Segmentation fault

anto, 2015-10-20:
I have tried the latest commits on the HEAD up to commit c000fc9e009f707b3ce8aa38449222e3f070343c, and I got "Segmentation fault". I built that on OpenWRT 14.07 build environment and the segmentation fault occurs when I ran BFGMiner on my TL-WDR3600 WiFi router.

In the mean time, I rolled back to the commit 517af40913ed356ac09a522e6ade56f9f32061eb and continue mining.

luke-jr, 2015-10-20:
Can you get a backtrace? What mining hardware?

anto, 2015-10-20:
Hi Luke,

How do I get the back trace? What tool do I have to use to get it?

I have only Antminer U3. It seems there are issues with probing the miner. I can start bfgminer only with --no-default-config option. But as soon as I enter the miner either "antminer:/dev/ttyUSB0" or "antminer:all", I always get Segmentation fault.
```
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
NO DEVICES FOUND: Press 'M' and '+' to add
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Enter "auto", "all", or a serial port to probe for mining devices.
Prefix by a driver name and colon to only probe a specific driver.
For example: erupter:/dev/ttyUSB39
Enter target:antminer:all
Segmentation fault
```

luke-jr, 2015-10-20:
```
gdb --args bfgminer <options>
run
<wait for crash>
bt
thr app all bt
```

luke-jr, 2015-10-20:
Think I found and fixed it, try latest git.

anto, 2015-10-20:
Thanks Luke,

I am still struggling now how to clean up the failed install of gdb and its libraries on my WiFi router as it is running out of space. I will compile your latest git after I manage to clean up the left over files.

luke-jr, 2015-10-20:
I think you'll need to opkg remove each part opkg install listed manually :(

anto, 2015-10-20:
I just compiled and installed you latest git up to commit 3f303439dd2ca3d329c7ac981e8d89cfc65438da. It runs fine now. So that confirms the segmentation fault issue is fixed.

luke-jr, 2015-10-20:
Great, thanks.

closed 2015-10-20 by luke-jr