luke-jr/bfgminer #642: Raspberry Pi 2 + HUB 10 ports + 8 units Avalon Nano 3 + BFGMINER 5.4 - Raspbian FREEZING

mobbil, 2015-11-10:
Hello im running the system as in subject after some seconds or minutes its freeze and crash the system, i have tried to make a log but is not recording the problem on it...

i have installed all packages, enabled icarus but i can find the problem is there someone can help me

Brgds

luke-jr, 2015-11-10:
Sounds like a hardware problem, and I have neither a RasPi2 nor any Avalon Nano to test with.

mobbil, 2015-11-10:
not a hardware problem i have two brand new rpi2 + 2 brand new rosewill hub10 ports... and 8 avalon i also tested in windows 8 with avalon installer and worked fine and also tested with just one avalon nano 3 peace

luke-jr, 2015-11-10:
Freezing/crashing the system is pretty much always a hardware or OS problem. Having multiples of the same hardware doesn't rule out bad design. RPi is notoriously known for having defective USB hardware; it wouldn't surprise me if RPi2 also has problems.

On that note, RPi2 is notorious for crashing if bright lights flash... is it possible all those miners have LEDs that could be triggering that?

mobbil, 2015-11-10:
ok i see your explanation but why works on windows? and i said i have multiples to show is going to be more difficult to have problems on all.. also when i install other miner its runs good for sometime but at certain moment i receive LIBUSB error
let me specify more infos for you to help me but the LEDs are red all the time.. untl crashes

Raspbian installed + apt-get update and upgrade + apt-get install packages + apt-get install bfgminer, autogen + configure enabled icarus + make 

im using this command
sudo ./bfgminer  -S ICA:/dev/ttyACM0  -o stratum+tcp://sha256.usa.nicehash.com:3334 -O btcaddress:x --set-device ICA:baud=115200 --set-device ICA:reopen=timeout --set-device ICA:work_division=1 --set-device ICA:fpga_count=1 --set-device ICA:probe_timeout=100 --set-device ICA:timing=0.22 --api-listen 

and after run im inserting one by one

both hub 10 ports and rpi2 are plugged in wall

any ideas?

Marko4200, 2016-01-12:
Hi, 

I am having similar problem, no expert but it does not seem like the drivers for this Avalon Nano 3 are not available in cgminer. I have purposely signed up to post tonight and would really appreciate any assistance. Do you need to add Avalon miner Driver?

Marko

Marko4200, 2016-01-12:
Sorry guys I have a rpi2 and Avalon nano and the device is not being picked up by bfg.

Marko

ChEeSyBoOfS, 2016-01-20:
I have BFGMiner 5.4.1 up and running a treat on a PI2 and it will run for hours, almost a day then the PI will lock up. About 12:05 to 12:10 in the attached syslog.

Any pointers would be greatly appreciated.

I have the PI plugged in with the official RPI mains adapter and the Avalon Nano 3 plugged into a powered USB hub.

If I pass the Avalon Nano 3 to a Windows 2012 VM over the network using VirtualHere and run BFGMiner 5.4.1 from under windows it doesn't lock up and will run for days. So the hardware checks out but I would just rather run BFGMiner directly from the PI not remotely via VirtualHere (installed on the PI) and windows.

Cheers

[dmesg.txt](https://github.com/luke-jr/bfgminer/files/97395/dmesg.txt)
[syslog.txt](https://github.com/luke-jr/bfgminer/files/97396/syslog.txt)