luke-jr/bfgminer #643: BMGMiner 5.4.0 dies on startup with Antminer U2

chazzTSC, 2015-11-13:
Was running 5.2.0-2 (special build created from 5.2.0 some months ago because of this same problem -- U2 dying at startup). Replaced with 5.4.0 and now crashes on startup again.

Command line bfgminer.exe --no-submit-stale --set-device antminer:clock=x0981 -o stratum+tcp://mint.bitminter.com:3333 -u chazz_antminer1 -p AnythingGoesHere -S amu:all - D -T --verbose

Log file attached

[bfgminer-log.txt](https://github.com/luke-jr/bfgminer/files/34364/bfgminer-log.txt)

luke-jr, 2015-11-13:
Where did your 5.2.0-2 come from?

chazzTSC, 2015-11-13:
You sent it to me yourself, on 9 July 2015. Issue #621 (?)

luke-jr, 2015-11-13:
#621 was about a hang, not a crash?

chazzTSC, 2015-11-13:
Ah. Yes, so it was. This one is a crash, even pops up the Windows 
"BFGMiner has stopped working" prompt. Windows 7 x64, 64-bit version of 
BFGMiner.

There was a second one, 5.2.0-2 crashes at startup when pool not found 
(626), but that was closed as no rep because I couldn't reliably 
replicate it, and seemed to stop being a problem when I switched to 
explicitly include stratum on the command line.

Probably I was conflating the two of them.

luke-jr, 2015-11-13:
Well, I'll need a backtrace of some sort to get anywhere on this... :/

chazzTSC, 2015-11-13:
Of course. Waiting for marching orders...

chazzTSC, 2015-11-13:
For what it's worth the antminer is coming up as com16.

luke-jr, 2015-11-14:
I don't know how to get a backtrace on Windows if it doesn't print it on the command prompt automatically.

chazzTSC, 2015-11-14:
While I can trigger the debugger on failure, I don't have a symbol file 
that I can use to make the resulting stack trace of any use. When you 
built the 64-bit Windows executable, it will have produced a .pdb file. 
Is there some place I can get that file?

luke-jr, 2015-11-14:
No, I don't think there is any way to produce a .pdb file with MingW. The official Windows binaries are all built with GDB-compatible debugging information in the EXEs/DLLs themselves.

luke-jr, 2015-11-14:
https://github.com/rainers/cv2pdb may work.

chazzTSC, 2015-11-14:
I'll take a look at that tomorrow; it's possible that one of the 
debuggers I have access to on that machine may be able to parse that.

chazzTSC, 2015-11-16:
My debugger doesn't seem to have picked up symbols, and as yet I have 
not been able to access my work desktop to build either gdb or the PDB 
conversion tool (both seem to be available only as source), but what I 
can see seems to match this back trace. I believe these are the same 
problem.

luke-jr, 2015-11-17:
Please both of you test http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=bfgminer to see if your problems are fixed.

chazzTSC, 2015-11-17:
Selected version 5.4.0-5-g7a98c1 seems to work properly on Windows 64 
with the Antminer U2.

dthorpe, 2015-11-17:
Build 5.4.0-5-g7a98c1 runs fine on Win7 32 bit.

Thanks!

closed 2015-11-17 by luke-jr