luke-jr/bfgminer #646: bfgminer keeps restarting through Macminer

turtwig59, 2015-12-08:
My bfgminer keeps resetting through macminer and I don't know why. This is a screenshot of it.
[image]
And another one.
[image]

Raginglikeaboss, 2015-12-13:
Same issue, using AMD cards as ell on OSX El Captain, any updates or things I'm doing wrong?

redskate, 2015-12-15:
Same issue with Yosemite iMac with "AMD Radeon R9 M290X Compute Engine"

fabulouspanda, 2016-01-12:
I can't reproduce this issue, this issue should mean that bfgminer crashed for some reason, but have you tried running it in Terminal without MacMiner i.e.

/Applications/MacMiner.app/Contents/Resources/bfgminer/bin/bfgminer

press enter and Terminal will prompt for pool, username and password

I also updated the binary I provide for bfgminer to 5.4.1 here
http://macminer.fabulouspanda.com/commandline/bfgminer/

Please let me know whether any of that helps and if so which bit in particular

PREngineer, 2016-06-21:
I get the same error.

After running bfgminer I get:

No devices found.  So, it seems it can't recognize my Radeon HD 7770 and that's why it fails.

I should point out... I'm using a hackintosh.  But my computer recognizes my card correctly.

This is interesting.

Thought this could help.

VytautasKuklys, 2017-08-29:
I know it's seems a old issue, but for me it's new and fabulouspanda comment on 12 Jan 2016 didn't solved my problem. But it helped to find out from terminal: "NO DEVICES" and I cannot add devices using "all" or "auto". I'm begginer, and want to use GPU pool mining for MFC (LTC clone) on the mac. Please help!
[image]

fabulouspanda, 2017-08-29:
It looks as though there are two different issues here, one where the device is found but then a crash occurs and one where no device is found. If the former I'd wonder whether perhaps the intensity is set to high but that's a complete guess. If the latter I've no idea why your cards aren't being found,  perhaps driver issues… Anyway, you'll need more information to troubleshoot, so run at the command line with debug output on. ./bfgminer --help is a good place to start