luke-jr/bfgminer #647: Set device sort order

VanessaE, 2015-12-13:
When bfgminer starts (and this is true for other mining programs), it queries the ASICs it's configured to use in a seemingly random order.  A given device (by serial number) might be GSD0 one time, but ends up assigned to, say, GSD6 another time.

It would be nice to be able to tell bfgminer to sort the active devices somehow, maybe just by serial number (I could see an argument for sorting by clock speed as well).

luke-jr, 2015-12-14:
They're queried *in parallel*, not any order at all. :)

Ordering them would serialise the process and make probing overall slower.

VanessaE, 2015-12-14:
I had a feeling that was actually the case.  Any chance the list could be sorted after they're queried? :)

luke-jr, 2015-12-14:
Not really, since they're (at least theoretically) added when they finish probing. To sort, it'd mean we must wait for all probes to complete before we can even begin using one that may have finished probing seconds earlier.

luke-jr, 2015-12-14:
Why do you think a sorted order is useful BTW? You can set properties by serial number...

VanessaE, 2015-12-14:
I figured it would make it a little easier to spot an underperforming (or dead) device with a quick glance at the list.

For example, in my current configuration, I have three gridseed blades and six "orbs", creating a scattered/unsorted list of twelve devices.  When one of the blades goes down, sometimes it gets lost among the lower-speed devices.  If it were grouped together with the others in the list, situations like that are easier to spot.

Plus it's more aesthetically pleasing to have them grouped together :)

luke-jr, 2015-12-14:
Well, sorting the *displayed list* by device type might be practical.