luke-jr/bfgminer #651: Solomining

PeterPdax, 2016-02-17:
I want to use solo mining with bfgminer but get Always an error.

   logfile:                  bfgminer: --generate-to: Invalid address

Is this a problem with bfgminer or my wallet or a syntax-error  ?
I am mining on gulden coin and its GBT supported.

My startfile:

bfgminer --scrypt -o http://localhost:9232 -u xxx -p yyy  --generate-to GPiN2At2Cc6VMS8WvpGuDj78rps3WyA1LZ    2>asolog.txt

luke-jr, 2016-02-17:
GBT is a very Bitcoin-specific protocol. It does not support altcoins.

buechling, 2016-06-19:
but you should be able to mine with getwork ( if i understood the difference right: http = getblocktemplate. no http = getwork )

this works for some altcoins: 
bfgminer --scrypt -o localhost:9232 -u xxx -p yyy

the wallet will choose a proper address

g1itch, 2017-08-14:
> the wallet will choose a proper address
New address for each block

adneerav, 2018-11-29:
Hello @luke-jr  

I am trying to do the same like @PeterPdax 

```
bfgminer -o http://192.168.39.103:2048 -u hellomyusername -p hellomypassword --generate-to WaAHv4bGsPcpKjYgJWLv9cn9Gm19qQUxU5 

 [2018-11-29 13:33:09] bfgminer: --generate-to: Invalid address
```

> **What else I tried ?**

```

bfgminer -o http://192.168.39.103:2048 -u hellomyusername -p hellomypassword

No servers could be used! Exiting.

```

**I also tried with localhost &  127.0.0.1**

**From the my altcoin cli i am able to get response of  `getblocktemplate` command as below**

**Source code fork from litecoin 0.15**

```
./wishcoin-cli getblocktemplate
{
  "capabilities": [
    "proposal"
  ],
  "version": 536870912,
  "rules": [
  ],
  "vbavailable": {
  },
  "vbrequired": 0,
  "previousblockhash": "f2f62bd49cf1a52f5e429e7adce94afb95c7370bfdb0cdc33be967df9d89dd66",
  "transactions": [
  ],
  "coinbaseaux": {
    "flags": ""
  },
  "coinbasevalue": 475600000000,
  "longpollid": "f2f62bd49cf1a52f5e429e7adce94afb95c7370bfdb0cdc33be967df9d89dd66270",
  "target": "00000ffff0000000000000000000000000000000000000000000000000000000",
  "mintime": 1543478592,
  "mutable": [
    "time", 
    "transactions", 
    "prevblock"
  ],
  "noncerange": "00000000ffffffff",
  "sigoplimit": 20000,
  "sizelimit": 1000000,
  "curtime": 1543478675,
  "bits": "1e0ffff0",
  "height": 277
}
```

**BFGminer Version Details**
```
 bfgminer --version
bfgminer 5.2.0
  Lowlevel: ftdi pci pci:uio pci:vfio spi usb vcom 
  Drivers: antminer avalon avalonmm bfx bifury bigpic bitforce bitforce:pci bitfury cairnsmore cointerra drillbit dualminer erupter gridseed hashbusterusb hashfast icarus klondike littlefury modminer opencl opencl:adl opencl:sensors proxy proxy:getwork proxy:stratum rockminer twinfury x6500 zeusminer ztex 
  Algorithms: SHA256d scrypt 
  Options: ncurses5 

```

Using cpuminer things are working fine its also using `getblocktemplate`

Please give your input if you can .
It would be great help to understand this issue if anyone can guide or has resolved this.

Thanks,

luke-jr, 2018-11-29:
GBT and BFGMiner are both Bitcoin-only.

closed 2018-11-29 by luke-jr

adneerav, 2018-11-29:
So what if I configure/build bgminer with `./configure --enable-scrypt` ?

Will BFGMiner support for scrypt ?

Reference link : https://github.com/litecoin-project/litecoin/issues/360#issuecomment-318515481

luke-jr, 2018-11-29:
Then BFGMiner will be compiled with support for hypothetical scrypt-mined Bitcoin sidechains and/or testnets.

jstefanop, 2018-11-29:
Gotta love your responses luke :D 

@adneerav yes bfgminer currently supports scrypt coins. Like luke said it does not support GBT for scrypt, its on my long to-do list but currently the workaround that works nicely is that you simply connect a stratum proxy to your scrypt coin's daemon, and just point bfgminer to the stratum proxy.

adneerav, 2018-11-29:
@jstefanop  @luke-jr  thanks for the response 

Let me try by setting up stratum and follow the process as you said.

Just asking will it work for solo mining too ? 

> Hello @luke-jr
>
> I am trying to do the same like @PeterPdax
>
> ```
> bfgminer -o http://192.168.39.103:2048 -u hellomyusername -p hellomypassword --generate-to WaAHv4bGsPcpKjYgJWLv9cn9Gm19qQUxU5 [...]

jstefanop, 2018-11-29:
Yes litecoind supports GBT but bfgminer GBT is only supported for bitcoind....GBT support for scrypt needs to be updated in bfgminer for it to work correctly.

Dextrac, 2020-11-16:
I am also trying to use a non-bitcoin address with the --generate-to option. When I try to run bfgminer with this option, it exits after printing "Invalid address" to the terminal emulator. bfgminer does in fact work with other sha256d coins like the one I am trying to mine. The fact that it will not accept a non-bitcoin address with the --generate-to option is completely arbitrary.