luke-jr/bfgminer #653: missing ./configure

lashworth, 2016-03-07:
I unzipped in to my linux directory to build, but ./configure is missing.. Not sure how to proceed. 

I have a number of zeus fury miners, are they usable with this program?

luke-jr, 2016-03-07:
Then you didn't download it from the right place.

closed 2016-03-07 by luke-jr

lashworth, 2016-03-07:
On 3/6/2016 10:58 PM, Luke-Jr wrote:
    
    
      
      Then you didn't download it from the right place.
      —
          it on GitHub.
      
        
          
          
        
        
      
    
    
      Thankyou for getting back to me so fast... Where is 'the right
      place?'
      Hate to be a pain, but I don't see where to go from here..
      
      Thanks!
    
  
 
 
		
		This email has been sent from a virus-free computer protected by Avast. www.avast.com