luke-jr/bfgminer #655: Bfgminer 5.4.1: JSON error

jeong760, 2016-03-24:
Hi

I've received a following error message during bitcoin solo mining using Bfgnminer 5.4.1 

 - SSM: JSON parse error: ?Version
 - Error from bufferevent

Regards,

luke-jr, 2016-03-24:
Something else did this.

closed 2016-03-24 by luke-jr

jeong760, 2016-03-24:
I was start with following command in order to do bitcoin solo mining with 2 of Antminer S3+ and Spondoolies SP20 miner on windows. 

after I ran with following command miner able to connect on but continuously received an error mesaged above. so I don't know how I solved this.

 - bfgminer --stratum-port=3333

And miner I was set following
 Stratum+tcp://192.168.1.5:333 
 Username
 Password

jeong760, 2016-03-24:
Dears,
 
I was updated what I did with BFGminer.
 
Regards,
John Ahn

luke-jr, 2016-03-24:
Note that solo mining with --stratum-port is *broken* in all released versions of BFGMiner, and only latest git is likely to work correctly. In any case, the error is a connection that doesn't do stratum correctly.

jeong760, 2016-03-24:
Ok, thank you. so what is correct methond in order to do solo mining with Bfgminer?
And only last git works correctly with --stratum-port option then how do I build for windows version?

luke-jr, 2016-03-24:
Generally you would run BFGMiner connected directly to the hardware (ie, on the controller).

You can get the latest git built from http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=bfgminer

jeong760, 2016-03-24:
Dears,
 
Thank you. 
and how I setup a solo mining configuration on bfgminer.conf file such as pool and connection port?
  
Regards,
John Ahn

jeong760, 2016-03-24:
Dears,
 
Following linked Bfgminer is support a  --stratum-port option? 
When I use that option then I've received an error message  that " SSM: No usable 2D work upstream!".
 
Could you give me a hint for http-port option of solo mining?
 
Regards,
John Ahn

jeong760, 2016-03-24:
Dears,

Following linked Bfgminer is support a  --stratum-port option? 
When I use that option then I've received an error message  that " SSM: No usable 2D work upstream!"

luke-jr, 2016-03-24:
Can you post a debug log?

```
bfgminer --debuglog --log-file debug.log --stratum-port 3333
```

jeong760, 2016-03-24:
Ok, please give me a time.

jeong760, 2016-03-24:
Dears,

Here I upload a log file but just I changed its extension name as txt because of Git does not support a log. 

[debug_20160324.txt](https://github.com/luke-jr/bfgminer/files/187414/debug_20160324.txt)

jeong760, 2016-03-24:
This is bfgminer version 5.4.1 that I download from bfgminer.org.  it works with --Stratum-port=3333 but still have an error message and miner goes to Sick mode. 

•SSM: JSON parse error: ?Version
•Error from bufferevent

jeong760, 2016-03-24:
This is bfgminer version 5.4.1 that I download from bfgminer.org. it works with --Stratum-port=3333 but still have an error message and miner goes to Sick mode.

•SSM: JSON parse error: ?Version
•Error from bufferevent

[debug.txt](https://github.com/luke-jr/bfgminer/files/187449/debug.txt)

jeong760, 2016-03-25:
Dears,

Any update result from the log files?

jeong760, 2016-03-25:
Dears,
 
I've already uploaded a log files for bfgminer with --stratum-port option. 
Could you guide me any solution?
 
I've fund that JSON parse error clue on Git but not sure this is related with bfgminer's behavior. 
 
  Link: https://github.com/Stratehm/stratum-proxy/commit/65d3afd619db4396dd93f3fb241caf913e3603f5
 
Regards,
John Ahn

luke-jr, 2016-03-25:
I see why it won't work for you. I need to wrap up a new version. Give me a few days.

luke-jr, 2016-03-25:
(Note this won't fix your original/main issue, though - the ?Version stuff is a problem with your miner)

jeong760, 2016-03-25:
Dears,
 
Thank you. I am surly wating for a new version.
So what is the main issue of current one?
 
Regards,
John Ahn

jeong760, 2016-03-25:
If the ?Version issuse is my miner problem then what kind of thing do you fixed on next version?

luke-jr, 2016-03-25:
The main issue is that working GBT->stratum support requires a newer version of libblkmaker, and is only compiled when the newer version is detected; but it is never detected, because the version number has not been bumped in libblkmaker yet (and the version used in the git submodule is too old anyway). So to finish fixing this, I need to complete the release of the new libblkmaker.

jeong760, 2016-03-25:
So, do you possible to solve it on next version?

luke-jr, 2016-03-25:
You can try http://luke.dashjr.org/tmp/code/webisect/webisect.php?dobuild=bfgminer again.

jeong760, 2016-03-25:
I was tried to run a bfgminer again, which downlaod form above link. but when after start it I've received an error. and I think above linked bfgminer already try to use before. 

 [2016-03-26 06:32:09] SSM: JSON parse error: {"params": ["jeong760", "56f5ae55-2", "1602", "V?e52", "766c90b7"], "id": 5404, "method": "minin
g.submit"}
 [2016-03-26 06:32:09] SSM: JSON parse error: {"params": ["jeong760", "56f5ae55-2", "1602", "V?e52", "92f88c1e"], "id": 5407, "method": "minin
g.submit"}

jeong760, 2016-03-25:
What does "Error from bufferevent" meaning? and why does it occurred during solo mining?

luke-jr, 2016-03-26:
> [2016-03-26 06:32:09] SSM: JSON parse error: {"params": ["jeong760", "56f5ae55-2", "1602", "**V?**e52", "766c90b7"], "id": 5404, "method": "mining.submit"}

This is garbage (see the part in bold). Old versions of cgminer (likely on your miner firmware) are known to have this problem. You'll need to either put BFGMiner on the firmware and use it directly, or at least upgrade the cgminer.

> What does "Error from bufferevent" meaning? and why does it occurred during solo mining?

It doesn't, unless your miner is broken.

jeong760, 2016-03-26:
I saw that you updated a libblkmake to 0.5.3 version. Does it solved a my issue with bfgminer? if so, I will compile it on windows version and test it again.

jeong760, 2016-03-26:
Antminer S series uses a Openwrt but I am not professional to import a BFGminer onto that firmware. 
Current cgminer version is 4.6.1-1.

jeong760, 2016-03-26:
Sopndoolies SP20 uses a cgminer 4.7.0 version but they've already discontinued this product so no more care about product.

jeong760, 2016-03-27:
I was update a cgminer 4.9.0 version on my ANTminers and now there is no Pasrse error. But dueing mining process miner speed is continuously downgrade. It was start as 800GB but 5 minutes later it will decrete as 100GHash in total. 

Does it related with bfgminer or my miner's settting issue?

luke-jr, 2016-03-27:
Maybe try running bfgminer with --set pxy:diff=128

Happy Easter!

jeong760, 2016-03-27:
Happy Easter too

jeong760, 2016-03-28:
If I disable a Longpoll using --no-long-pool option of bfgmeiner then what will affect? And what is the long-pool?

luke-jr, 2016-03-28:
Longpolling is an optimisation. It has no practical use for disabling, just testing.

jeong760, 2016-03-28:
How I express thank you?

luke-jr, 2016-03-28:
Please feel free to make a donation to 12hzAYYxm8fTEGRPCRYfSLuk3H5Uwzyx7h if you like.

jeong760, 2016-03-28:
Do you accept PayPal?

jeong760, 2016-03-28:
Following logs are meaning that my miner work for solo mining againest to my personal wallet software. and I was placed my BTC addres as BTC address for just in case.
Am I correct?

[2016-03-28 15:53:39] Generated stratum header 00000004dc324c07650c5a0d368bb9bfe87415bda8dc4ce7057059aa0000000000000000c6ebb3931bf23b8567efe85f1feebb44b107f18e9784c6a1a59020986c58354a56f8d4f11806a4c300000000

[2016-03-28 15:53:39] Work job_id (null) nonce2 bbff8a16
[2016-03-28 15:53:39]  Proof: 00000000009879ac73f4e26d551b401ded650142112928e0cd27fa08b293315d
Target: 000000000000000006a4c3000000000000000000000000000000000000000000

TrgVal? no (false positive; hash > target)
[2016-03-28 15:53:40] SSM: RECV: {"params": ["BTC address", "56f8d4d3-f4", "9c17", "56f8d4d2", "1cd417d2"], "id": 12060, "method": "mining.submit"}
 
[2016-03-28 15:53:40] Generated stratum header 00000004dc324c07650c5a0d368bb9bfe87415bda8dc4ce7057059aa0000000000000000696f67f1609334eb3775e8301e352acfc5b74dfb399d650d27b6165f03fa1f8f56f8d4f21806a4c300000000

[2016-03-28 15:53:40] Work job_id (null) nonce2 bbfe9c17
[2016-03-28 15:53:40]  Proof: 000000000029a887acd76b90de9ca8819e03d225506d9a0ccfb52e9f83eca093
Target: 000000000000000006a4c3000000000000000000000000000000000000000000

[2016-03-28 15:51:49] Successfully rolled extranonce to dataid 9
[2016-03-28 15:51:49] Generated work from latest GBT job in get_work_thread with 32767 seconds left
[2016-03-28 15:51:49] Pushing work 56605 from pool 0 to hash queue
 
[2016-03-28 15:51:49] Selecting pool 0 for work
[2016-03-28 15:51:49] Successfully rolled extranonce to dataid 10
[2016-03-28 15:51:49] Generated work from latest GBT job in get_work_thread with 32767 seconds left

luke-jr, 2016-03-28:
If you would like to donate via PayPal, please use luke+paypal@dashjr.org for the email address.

The log looks okay, but note that the stratum username ("BTC address" in the log) is ignored entirely.

jeong760, 2016-03-28:
Luke-jr, you PayPal account is not vaild. please check and leave here correct one.

jeong760, 2016-03-28:
if stratum is inore my BTC address then how I get a BTC when found? it uses a coin base address, which generate by bfgminer when it start. am I correct?

luke-jr, 2016-03-28:
It generates directly to your wallet, using an address your wallet gives it. Or you can use --generate-to ADDRESS to override this.

For PayPal, that is the correct account I login with. You can also try http://paypal.me/lukedashjr

jeong760, 2016-03-28:
PayPal.me is not availble in my country. let me know your vaild account.

luke-jr, 2016-03-28:
Oh well, I guess PayPal is broken then. luke+paypal@dashjr.org is correct.

jeong760, 2016-03-28:
Sent.. please check with PayPal account. and I want to lean about Eloipool setup for my personal mining of bitcoin.. 

Anyway thank you and appreciate you.

John Ahn

luke-jr, 2016-03-28:
Thanks. Note that Eloipool does not support Windows, so you will need a Linux system to set it up.

jeong760, 2016-03-28:
Yes, I know and I am famlier deal with Linux OS. I found that following link but is this need to installation a MySQL as its data store location? 

Link: https://github.com/luke-jr/eloipool/issues/3

What I did

- Installation a Ubuntu 14.04LTS
- Installation a bitcoind and bitcoin-qt and finished a block sync
- installation a MySQL and its GUI management program

I wat to know exact installation guide for Eloipool but only I found that above link.

Do you have any good guide for it?

luke-jr, 2016-03-29:
No, I don't have any guide. If it's just for you, you can just disable share logging with ShareLogging=()
Then no MySQL is required at all.

jeong760, 2016-03-29:
By the way how I download a midstate

luke-jr, 2016-03-29:
You don't need it.

jeong760, 2016-03-29:
But midstate is mandatory option for The Eloipool setup?

luke-jr, 2016-03-29:
No, it works fine without it.

jeong760, 2016-03-29:
Could you give me download link for midstate? Http://gitorious.org/midstate/midstate is dead.

luke-jr, 2016-03-29:
Again, you literally do not need it. Every miner since 2012 does the midstate on its own.

jeong760, 2016-03-29:
Ok, so I will go forward to make my own pool.

Thank you