luke-jr/bfgminer #657: Solomining issue.

jeong760, 2016-04-08:
Dears,

I want to solo mining wiht bitcoin-qt of my local but I cannot so please check and reply me. 

Bitcoin.conf
 server=1
 port=8333
 rpcuser=bitcoinrpc
 rpcpassword=bitcoinpassword
 rpcport=9332

The setting values of run_bfgminer.bat
bfgminer -o http://localhost:9332 -u bitcoinrpc -p bitcoinrpcpassword --stratum-port=9332 -net-delay --no-report-stale --no-hotplug --request-diff=1.5 --queue=8192

after I saw that Pool 0 but displayed an error SSM No suitable 2D works. so I think Bfgminer has an issue with Stratum setting or solo mining option.

0:Enabled   Strtum  Quota 0 Pool 0:http://localhost:9332  User:bitcoinrpc
1: Enabled   GBT    Quota 0 Pool 0: http://localhost:8332/#allblocks User:bitcoinrpc

Regards,
John Ahn

luke-jr, 2016-04-08:
5.4.2?

jeong760, 2016-04-08:
Yes, version is 5.4.2

jeong760, 2016-04-08:
Screen is displaying as follow and it makes a ghost PXY0 device on the list.

 PXY 0:            0.0/  0.0/  0.0 h/s │ A:0 R:0+0(none) HW:0/non
───────────────┴───────────────────────┴──────────────────────
2016-04-08 12:32:34] Waiting for devices; press 'M+' to add, or 'Q' to quit
 [2016-04-08 12:32:34] Added local bitcoin RPC server on port 8332 as pool 1
 [2016-04-08 12:32:34] Probing for an alive pool
 [2016-04-08 12:32:34] Network difficulty changed to 167G ( 1.19Eh)
 [2016-04-08 12:32:34] Switching to pool 1 http://localhost:8332/#allblocks - first alive pool
 [2016-04-08 12:32:34] Long-polling activated for http://localhost:8332/#allblocks (getblocktemplate)
 [2016-04-08 12:32:36] Pool 0 http://localhost:9332 alive, testing stability
 [2016-04-08 12:32:36] Pool 0 http://localhost:9332 alive
 [2016-04-08 12:32:36] Switching to pool 0 http://localhost:9332
 [2016-04-08 12:32:36] Pool 0 is hiding block contents from us
 [2016-04-08 12:32:36] SSM: No usable 2D work upstream!
 [2016-04-08 12:32:36] Message from pool 0: Current upstream pool does not have usable 2D work
 [2016-04-08 12:32:36] Stratum connection to pool 0 interrupted
 [2016-04-08 12:32:36] SSM: No usable 2D work upstream!

[SOLO1_Debug - copy.txt](https://github.com/luke-jr/bfgminer/files/209602/SOLO1_Debug.-.copy.txt)

jeong760, 2016-04-08:
[2016-04-08 12:34:42] PXY 0: Idle for more than 60 seconds, declaring SICK!
 [2016-04-08 12:34:42] Pool 0 http://localhost:9332 alive, testing stability

luke-jr, 2016-04-08:
Get rid of the -o, -u, and -p options. You're causing BFGMiner to run in a loop. >_<

jeong760, 2016-04-08:
Dears,
 
Ok, I will remove them but it is described on readme.txt file. 
and during solo mining there is no displayed on Accept and Reject and then HW error count.  is that normal?
 
 PXY 0:        │497.0/438.4/438.4G h/s │ A:0 R:0+0(none) HW:0/no
 PXY 1:        │416.0/465.7/465.78Gh/s │ A:0 R:0+0(none) HW:0/none
 PXY 2:          467.8/465.1/465.1Gh/s │ A:0 R:0+0(none) HW:0/non
 PXY 3:          443.8/436.2/436.2Gh/s │ A:0 R:0+0(none) HW:0/non
 PXY 4:           1.15/ 1.14/ 1.14Th/s │ A:0 R:0+0(none) HW:0/non
 
 
Regards,
John AHn

luke-jr, 2016-04-08:
Yes, an accept would be a block found.

jeong760, 2016-04-11:
Dears,
I've another question as following during stratum mining against my local wallet of bitcoin-qt.
 why work job_id is displaying as null?
 
[2016-04-11 10:21:02] Work job_id (null) nonce2 bbfd5f13
[2016-04-11 10:21:02]  Proof: 000000000030653911926905c99cbdc2b308c95442ca1b26a76877d834f6e170
Target: 00000000000000000696f4000000000000000000000000000000000000000000 
 
Regards,
John Ahn

luke-jr, 2016-04-11:
That's debug info. That particular job_id is only used for stratum pools, not GBT.

jeong760, 2016-04-11:
Dears,
 
So during solomining it does not use. am I correct? 
But in order to solo mining against local wallet miner should be uses a stratum-port option. so I wonder why it does not display as stratum+GBT on bfgminer. 
 
Regards,
John Ahn