luke-jr/bfgminer #658: The inquiry of errror message from BFGMienr.

jeong760, 2016-04-19:
Dears,

I've recevied an following message on BFGminer but I am not sure this is related with BFGminer 
however I want know the reason why following errors occurred.

 - I use a Eloipool as solo mining pool fo bitcoin
 - miner is connect to Eloipool using BFGMiner 

- Error message are
 - "Method 'mining.suggest_target' not found", null
 - Message from pool 0: This stratum server is licensed under the GNU Affero General Public 
   License, version 3. You may download source code over stratum using the server.get_source
   method.

Regards,
John Ahn

luke-jr, 2016-04-19:
> "Method 'mining.suggest_target' not found", null

That looks like a mere warning?

> Message from pool 0: This stratum server is licensed under the GNU Affero General Public License, version 3. You may download source code over stratum using the server.get_source method.

Copyright disclosure. Not an error.

closed 2016-04-19 by luke-jr

jeong760, 2016-04-19:
How I can check for Eloipool and BFGMiner working well together?