luke-jr/bfgminer #659: Can not set up config

RepoBike, 2016-04-20:
Trying to config before installing for cpu mining LTC. Or atleast trying to. So far nothing has been working.

Linux Mint 17.2

This is the 6th guide iv found and this is the 6th time it has failed to install. 
Might need a way better install guide. 
Installing from the software MGR works but nothing enabled and unable to edit the settings.

any other options out there that will work?

luke-jr, 2016-04-20:
Altcoins are not officially supported.

closed 2016-04-20 by luke-jr

RepoBike, 2016-04-20:
Thank you for the reply.

Whats a good miner for LTC?

luke-jr, 2016-04-20:
I also personally don't support [scams like Litecoin](https://bitcointalk.org/index.php?topic=23768.msg3815348#msg3815348).