luke-jr/bfgminer #661: bfgminer as a daemon : fails to contact pool during boot

MoonSweep, 2016-05-16:
Hi,

I'm trying to modify Debian's bfgminer package to make it run as a daemon. I wrote a [patch](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=824117) including amongst other things, an init script that relies on Debian's start-stop-daemon utility to fork. It works flawlessly when manually run after boot, but not during boot. It's extremely weird.

BFGMiner exits after saying that it can't contact the pool:

````
 [2016-05-16 18:35:13] Probing for an alive pool
 [2016-05-16 18:35:14] No servers were found that could be used to get work from.
 [2016-05-16 18:35:14] Please check the details from the list below of the servers you have input
 [2016-05-16 18:35:14] Most likely you have input the wrong URL, forgotten to add a port, or have not set up workers
 [2016-05-16 18:35:14] Pool: 0  URL: stratum+tcp://stratum.mining.eligius.st:3334  <credentials>
````

The debug output doesn't say anything more on this matter (it only adds information about hardware detection).

It's obviously not a misconfiguration problem with the pool URL/credentials since it works when run after boot. I'm pretty sure it's not a network access problem either, because during my numerous tests I tried to telnet the port on the pool host from withing the init script, and it worked. A lagging/slow pool without a backup could be a culprit too, but if BFGMiner wasn't happy with such a pool, the init script shouldn't work after boot either, right ? However, it does. I'm at a loss here.

So my question is: what could make BFGMiner fail in this case ?

Regards,

-- 
Raphaël Halimi

luke-jr, 2016-05-16:
Maybe name resolution/DNS isn't ready yet?

MoonSweep, 2016-05-16:
Le 16/05/2016 20:12, Luke-Jr a écrit :
> Maybe name resolution/DNS isn't ready yet?

(thanks for answering so quickly)

A DNS failure is unlikely, since that would have made the telnet test
fail, too.

Any other idea ?

-- 
Raphaël Halimi

luke-jr, 2016-05-17:
Not sure, sorry. If you run it in strace, does it reveal anything interesting?

MoonSweep, 2016-05-17:
> Not sure, sorry. If you run it in strace, does it reveal anything
> interesting?

Sorry to bother you with this... I managed to trace the binary
(including forks) through the init script, but since I'm an admin, I
speak bash and POSIX shell fluently, but unfortunately not C, and thus I
don't understand the resulting file at all... I tried to read it, but I
didn't find anything relevant between the "Probing..." and "No
servers..." write to STDOUT/STDERR calls.

May I please show it to you ?

Regards,

-- 
Raphaël Halimi

luke-jr, 2016-05-17:
Sure, just post a link here.

MoonSweep, 2016-05-17:
Does it contain any sensitive information ? I didn't find my address in it, but I'm not sure there's nothing I missed.

luke-jr, 2016-05-17:
If you would prefer to email me a private link, feel free to send the link (no attachments please) to luke+bfgminer+issue661@dashjr.org

MoonSweep, 2016-05-29:
Ping ?

wudc, 2018-06-08:
Hi I setup the bfgminer bit 32 version for my Windows 10 desktop with Nivida graphic card. I also setup an Antpool account with a wallet. But when I startup the bfgminer against the antpool url: stratum+tcp://stratum.antpool.com:3333, then userid.worker and password (my antpool account information), I immediately got this error: Pool setup failed. Where do I get more error details? I checked my bfgminer folder and couldn't find any log or error file. Please help.

wudc, 2018-06-08:
I also have my Norton security running. Could it be Norton is blocking my connection to the Antpool server?