luke-jr/bfgminer #668: Using bfgminer 5.4.2 with Antminer U2+ only generates "Comms error werr=1" messages on the console

swtrse, 2016-06-28:
I am using bfgminer 5.1.0 which is working flawless with my Antminer U2+ devices.
I decided to upgrade to version 5.4.2 so I downloaded the windows build for 64bit and made a test run.
All settings are in the config file so I call bfgminer without parameters. I let the config file untouched and started the new version.
That version seems not to work with AntMinter U2+ devices at all.
The Console is flooded with "Comms error werr=1" and no accepted shares are shown.

So what have changed?

luke-jr, 2016-06-28:
Please follow the instructions on http://luke.dashjr.org/tmp/code/webisect/webisect.php?session=pt668 to determine when exactly this broke. Note that this produces intermediate builds of BFGMiner *between* versions to test, so while it may seem like you're testing the same versions in some cases, they *are* different. You'll probably need to test 7 or 8 different versions before it narrows down the problem. When it's done giving you stuff to test, just post the outcome here.

swtrse, 2016-06-28:
I would but in the middle of the process your site stopped responding. I will try it tomorrow an post the outcome

iDunk5400, 2016-06-28:
No problems here with 5.4.2-7-g9f781b8 and my Antminer U2+ @2.40 Gh/s.

swtrse, 2016-06-29:
@luke-jr This is the output

e75cce8f063059c1b3b1b92d90436ab87fad5a64 is the first bad commit
commit e75cce8f063059c1b3b1b92d90436ab87fad5a64
Author: Luke Dashjr 
Date:   Thu Jul 9 08:17:37 2015 +0000

    Bugfix: icarus: Never set timeout to 0, since it disables the timeout altogether

:100644 100644 34d6f347c3727e6db65ac60f142dd06ae64651b5 e93f541b5859131982eef35e1608bef45c41fe5c M	driver-icarus.c
:100644 100644 28d9ae5a5591c313d3ad6e3094b51c91b181b8e9 7d4f7f7531cd5be75cdbf25221c9b83b2b34a3a2 M	lowl-vcom.h

@iDunk5400 what are your settings?
"api-allow" : "W:127.0.0.1",
"api-listen" : true,
"api-mcast-port" : "4028",
"api-port" : "4028",
"expiry" : "120",
"log" : "5",
"queue" : "4",
"scan-time" : "60",
"shares" : "0",
"gpu-threads" : "2",
"gpu-platform" : "0",
"temp-hysteresis" : "3",
"skip-security-checks" : "0",
"expiry-lp" : "3600",
"scan" : "all",
"no-client-reconnect" : true,
"failover-only" : true,
"failover-switch-delay" : "300",
"set-device" : "NFY:osc6_bits=52",
"no-pool-disable" : true,
"no-show-processors" : true,
"no-show-procs" : true,
"no-unicode" : true,
"quiet-work-updates" : true,
"quiet-work-update" : true,
"submit-stale" : true

To be complete I have attached 6 Antminer U2+ and 1 IceFurry

iDunk5400, 2016-06-29:
I only have these that are relevant to the U2+:
-S antminer:all
--set-device antminer:clock=x0B81
--set-device antminer:timing=short

I have a single U2+ attached :)

luke-jr, 2016-06-29:
@swtrse That suggests your timeout setting for the device is too low. What are you using for device config?

swtrse, 2016-06-30:
This are the Settings in my config file (without the pools) I do not use any other settings just starting the .exe
As stated above I use 6 Antminer U2+ and one IceFury
`"api-listen" : true,
"api-mcast-port" : "4028",
"api-port" : "4028",
"expiry" : "120",
"expiry-lp" : "3600",
"failover-only" : true,
"failover-switch-delay" : "3600",
"gpu-platform" : "0",
"log" : "5",
"no-adl" : true,
"no-pool-disable" : true,
"no-client-reconnect" : true,
"no-show-processors" : true,
"no-show-procs" : true,
"no-unicode" : true,
"queue" : "4",
"quiet-work-updates" : true,
"quiet-work-update" : true,
"scan-time" : "60",
"skip-security-checks" : "0",
"submit-stale" : true,
"temp-hysteresis" : "3",
"shares" : 0,
"kernel-path" : "C:\\Program Files\\BFGMiner\\/share/bfgminer",
"scan" : [
	"all"
],
"set-device" : [
	"NFY:osc6_bits=52"
],
"api-allow" : "W:127.0.0.1"`