luke-jr/bfgminer #669: bfgminer crashes when adding a device.

wsanders99, 2016-07-02:
bfgminer version 5.4.2-unknown
Installed with brew tap nwoolls/xgminer
then brew install --HEAD bfgminer

using the command bfgminer -o http://mint.bitminter.com:3333 -u username -p password -I d
this brings up the main interface and I am prompted to add a device.
pressing m and + then prompts my to type _all, auto or a serial port_ 
upon entering auto, the command line outputs this:

[1] 49468 trace trap [the command I used earlier]

luke-jr, 2016-07-02:
Can you get a backtrace, or at least a debug log?

wsanders99, 2016-07-02:
Kind of new to the debug thing, but I think I did this right?

`Starting program: /usr/local/bin/bfgminer 
warning: `/BinaryCache/coreTLS/coreTLS-35.40.1~1/Objects/coretls.build/coretls.build/Objects-normal/x86_64/system_coretls_vers.o': can't open to read symbols: No such file or directory.
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_ciphersuites.a"
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_handshake.a"
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_record.a"
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_stream_parser.a"
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/bdz.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/bdz_ph.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/bmz.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/bmz8.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/brz.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/buffer_entry.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/buffer_manager.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/chd.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/chd_ph.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/chm.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/cmph.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/cmph_structs.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/compressed_rank.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/compressed_seq.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/fch.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/fch_buckets.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/graph.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/hash.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/jenkins_hash.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/linear_string_map.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/miller_rabin.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/select.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/vqueue.o': can't open to read symbols: No such file or directory.
warning: `/BinaryCache/cmph/cmph-1~1091/Objects/cmph.build/cmph.build/Objects-normal/x86_64/vstack.o': can't open to read symbols: No such file or directory.

warning: `/Volumes/Builds/nv/build/apple/rel/Syrah/NVSyrah-346-02-03/drivers/apple-pvt/drvNVIDIA/Common/obj/GLRenderer/Release/Web/gpgpu/built/x86_64_Darwin_release/cuda/src/cuda_driver/traphandler_abi_cilp.cubins.o': can't open to read symbols: No such file or directory.
warning: `/Volumes/Builds/nv/build/apple/rel/Syrah/NVSyrah-346-02-03/drivers/apple-pvt/drvNVIDIA/Common/obj/GLRenderer/Release/Web/gpgpu/built/x86_64_Darwin_release/cuda/src/cuda_driver/traphandler_kilp.cubins.o': can't open to read symbols: No such file or directory.
warning: `/Volumes/Builds/nv/build/apple/rel/Syrah/NVSyrah-346-02-03/drivers/apple-pvt/drvNVIDIA/Common/obj/GLRenderer/Release/Web/gpgpu/built/x86_64_Darwin_release/cuda/src/cuda_driver/unittest.o': can't open to read symbols: No such file or directory.
[... cut ...]
                                    [New Thread 0x1c03 of process 53881]
                                                                        [New Thread 0x1d03 of process 53881]

                                                                                                            Thread 12 received signal SIGTRAP, Trace/breakpoint trap.
                                                      [Switching to Thread 0x1b33 of process 53881]
0x00007fff8b40b5b4 in CFRetain ()
   from /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation`

luke-jr, 2016-07-02:
Sounds like buggy GPU drivers. Maybe try adding the specific device(s) you want explicitly, so you don't add GPUs?

closed 2016-07-02 by wsanders99