luke-jr/bfgminer #672: ktmpl error: Unrecognized block version, and not allowed to reduce or force it

jeong760, 2016-07-14:
Dears,

Yesterday I've try to start a solo mining with Bfgminer5.4.2 but I've got following error. 
I think that bfgmienr does not recgnized a bitcoin block version after half count event on July 10.

[2016-07-14 10:24:49] Started bfgminer 5.4.2
 [2016-07-14 10:24:50] No devices detected!
 [2016-07-14 10:24:50] Waiting for devices; press 'M+' to add, or 'Q' to quit
 [2016-07-14 10:24:50] Testing pool http://localhost:8332/#getcbaddr#allblocks
 [2016-07-14 10:24:50] blktmpl error: Unrecognized block version, and not allowed to reduce or force it
 [2016-07-14 10:24:50] JSON decode failed(1): '[' or '{' expected near 'HTTP'
 [2016-07-14 10:24:50] JSON decode failed(1): '[' or '{' expected near 'HTTP'

Regards,
John Ahn

NainKult, 2016-08-13:
I have the same issue here

luke-jr, 2016-08-13:
What node software are you using? Bitcoin Core 0.12.1 has a known bug that would cause this, so you need either Core 0.13.0rc3, or [Knots](http://bitcoinknots.org/) 0.12.1.

NainKult, 2016-08-14:
I'm using [0.12.1-bitcore](https://github.com/bitpay/bitcoin) for dev purposes and i have an old miner which i use for lottery (and heating the cat bed)

I'm stuck with 0.12.1 until they merge to 0.13, which i assume they'll wait for final version to come out before releasing.

Is there a patch who address this particular issue ?

luke-jr, 2016-08-14:
https://patch-diff.githubusercontent.com/raw/bitcoin/bitcoin/pull/8176.diff

NainKult, 2016-08-14:
`bitcore@dev002:~/bitcoin$ patch -p1 < 8176.diff
patching file qa/rpc-tests/bip9-softforks.py
Hunk #2 succeeded at 94 with fuzz 1.
patching file src/consensus/params.h
patching file src/main.cpp
Hunk #1 succeeded at 2301 (offset 164 lines).
patching file src/main.h
Hunk #1 succeeded at 822 (offset 290 lines).
patching file src/rpcmining.cpp
patching file src/versionbits.cpp
patching file src/versionbits.h`

Pool 0 http://localhost:8332 alive

Work like a charm, thanks again :+1:

luke-jr, 2016-08-14:
@jeong760 If your problem is different, let me know and we can reopen this...

closed 2016-08-14 by luke-jr

jeong760, 2016-08-16:
Hi Luke-jr,
 
I am using a windows based client. 
so how I appiled a patch of windows bitcoin core?
 
Regards,
John Ahn

luke-jr, 2016-08-16:
@jeong760 There is no simple way to patch a Windows node. You'll just need to use one without the bug.