luke-jr/bfgminer #674: Incorrect display average hash rate with Zeus (RusMiner60)

kr1z1s, 2016-07-22:
Hi.

With my Zeus device (RusMiner60) not displaying correctly 20 second exponentially decaying average hash rate and all time average hash.
All time average hash rate based on actual nonces found, adjusted for pool  reject and stale rate - correct.

[image]

jstefanop, 2016-07-22:
Which variant of Zeus Miner is it (specifically how many chips does it have)? Looks like the driver is using the wrong number of chips.

kr1z1s, 2016-07-25:
I have device with 20 chip Inosilicon A2. Average hash rate ~ 60 Mh.
Option "--set zeusminer:chips=" i tried to change. The result as in the screenshot. I did not succeed to achieve the real values.

But in [this fork](https://github.com/Darkwinde/bfgminer) all displayed hashrate - correct. There is a different driver for Zeus.