luke-jr/bfgminer #675: configure: error: Package requirements (libevent >= 2.0.3) were not met:

waldyd, 2016-07-31:
Hello,

While building bfgminer, the following was prompted:

`configure: error: Package requirements (libevent >= 2.0.3) were not met:`

But the last estable versión is:

```
[waldyd@master bfgminer]$ dnf list libevent
Last metadata expiration check performed 0:13:51 ago on Sat Jul 30 21:01:15 2016.
Installed Packages
libevent.x86_64                                                                                              2.0.21-6.fc22                                                                                               @anaconda
```

Any ideas about how to solve this issue?

Best Regards,

Néstor Waldyd

luke-jr, 2016-07-31:
Sounds like a broken distro/install, but just in case can you upload your config.log?

Also, libevent should only be required for stratum proxy support - is configure  refusing to proceed without it?

waldyd, 2016-07-31:
Hello,

Thanks for your answer.

Regarding your configure related question: yes, configure is refusing to proceed:

```
checking for nanosleep... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for JANSSON... yes
checking for libmicrohttpd... yes
checking for libevent... no
configure: error: Package requirements (libevent >= 2.0.3) were not met:

No package 'libevent' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libevent_CFLAGS
and libevent_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

Attached it is my config.log file.

**After installing the libevent-devel package, the problem was solved.**

Thanks again for your help.

Best Regards,

Néstor Waldyd

[config.log.zip](https://github.com/luke-jr/bfgminer/files/392495/config.log.zip)

anilpai, 2016-08-22:
I dont think its a bug... You just need to install the dependencies.

If you are on Debian based OS, just do 

`sudo apt-get install libevent-dev`

oliveiraped, 2017-04-05:
Yes That's it works now !!!

closed 2017-04-06 by luke-jr

BasicPhreak, 2017-11-10:
I am having this problem too.
I try to do `sudo apt-get install libevent-dev`
and get these results: 
`sudo apt-get install libevent-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libevent-core-2.1-6 libevent-extra-2.1-6 libevent-openssl-2.1-6 libevent-pthreads-2.1-6
The following NEW packages will be installed:
  libevent-core-2.1-6 libevent-dev libevent-extra-2.1-6 libevent-openssl-2.1-6 libevent-pthreads-2.1-6
0 upgraded, 5 newly installed, 0 to remove and 1 not upgraded.
Need to get 456 kB of archives.
After this operation, 2,147 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Abort.`
Everytime I say Yes it Aborts. I am fairly new to linux so please keep that in mind with your replies.

lanaparadinha, 2017-12-15:
Thx anilpai!