luke-jr/bfgminer #676: JSON decode failed(1): string or '}' expected near 'id'

waldyd, 2016-07-31:
Hello,

While mining, i am having the following warning:

```
[2016-07-31 12:39:57] Stratum from pool 0 detected new block
 [2016-07-31 12:39:57] JSON decode failed(1): string or '}' expected near 'id'
 [2016-07-31 12:39:57] JSON decode failed(1): string or '}' expected near 'id'
 [2016-07-31 12:39:57] Unknown stratum msg: { id: null, result: [], error: true }
```

Would you please give me any ideas to solve this issue?

Best Regards,

Néstor Waldyd

luke-jr, 2016-07-31:
Looks like a broken pool. JSON keys need to be quoted.

waldyd, 2016-07-31:
Hello, 

Thanks for your answer.

I am using GHash.io pool. May this be due to other reason?

Best Regards,

Néstor Waldyd

luke-jr, 2016-07-31:
Can you email me the full config including login info? (only login for mining, NOT for accessing the account) This is so I can run it myself and debug.

waldyd, 2016-07-31:
Done! (luke+github_public@dashjr.org)

luke-jr, 2016-07-31:
Looks like this is ghash.io's bug. It should be safe to ignore though.

closed 2016-07-31 by luke-jr