luke-jr/bfgminer #685: solo mining with bitcoin-core v0.13.1: "method (getwork) not found"

hpbock, 2016-12-20:
When I try to mine on testnet using bitcoin-core v0.13.1 and the current segwit branch of bfgminer, my bitcoin-core is reported as dead:

    1: Enabled  Dead  Quota 1 Pool 1: http://mc:18332  User:bitcoin

I traced the tcp connection and stumbled upon one connection that looks strange to me. At first bfgminer makes a json/rpc request to "getblocktemplate" which bitcoin responds to. Then bfgminer makes a request to the method "getwork" which bitcoin has not implemented.

Why does bfgminer call a method that is not implemented in bitcoin-core?

# request to getblocktemplate
> POST / HTTP/1.1
> Host: mc:18332
> Accept: */*
> Accept-Encoding: deflate, gzip
> Content-type: application/json
> X-Mining-Extensions: longpoll midstate rollntime submitold
> X-Mining-Hashrate: 2026000000
> Content-Length: 272
> User-Agent: bfgminer/5.4.2-17-g4f861b4
> 
> {"id": 0, "method": "getblocktemplate", "params": [{"rules": ["csv", "segwit"], "capabilities": ["coinbasetxn", "workid", "longpoll", "coinbase/append", "time/increment", "version/force", "version/reduce", "submit/coinbase", "submit/truncate"], "maxversion": 1073741823}]}

# response 
> HTTP/1.1 200 OK
> Content-Type: application/json
> Date: Di, 20 Dez 2016 19:58:32 GMT
> Content-Length: 5994
> 
> {"result":{"capabilities":["proposal"],"version":536870912,"rules":["csv","!segwit"],"vbavailable":{},"vbrequired":0,"previousblockhash":"0000000000343071a5931b82c73966352a353bcac08c507da4567829fb4464a8","transactions":[{"data":"0100000001104e170496ad958544dfc6cfef67a657b246ce062645f56b92d5a60922c513f5010000006b483045022100ebc8be3c41676d7882411f7018f44939aca072cbe168093afc571cf6a0b2fd6502206442c8cb78db3014cde79216d9a256200c3e3e6017b9a9af1d716a3a98d2b4e80121037eb170dd5dff1a18964a9bc969855d41dcf8f3711a9a50894891ddbc2251ea83ffffffff02a0118721000000001976a914e1e149c03155884e4a01abac2da7f000b1f7fb7688ac754961b0150000001976a91411ca13ec984de5ec05490d121136a44f6de721c588ac00000000","txid":"956abc889e83518072105eaf254fb6fcbcdacda220859e9ce19ed28f21e2f222","hash":"956abc889e83518072105eaf254fb6fcbcdacda220859e9ce19ed28f21e2f222","depends":[],"fee":100000,"sigops":8,"weight":904},{"data":"0100000001e19f327b3d15d7f32df5a960994846da8a20dbb4262e4f12a57b570131c836e6020000006a47304402200d65f1bfad9e26ad50c28cd09761d7614cb9b510503a91ed950295ec223c776902201e5c7386af57727870d8f54112858074b5e5af72e701403100b4e1847108b3260121035a66a61829921a8fec9da92fabc2310cc8338fdb236b6cc422f02abde842a748ffffffff0358020000000000001976a914d4b04931eef26e09791d68a3ba153fd467a3ebf288ac00000000000000003e6a3c43430201ce7d53f4fba65972b91f368729eb83a5fbb30c5e2972d9268dd8898950dd3a790e63c80a0fd3c0b04a563fe1413f0c424d5addc50100011050868521000000001976a914ac35c29c6f7687bb329e42b826c5694f49352d0388ac00000000","txid":"b67da210f2ad807eeed784f1d88d4d861fd90f9e8fc12901b9899f37b95bf55d","hash":"b67da210f2ad807eeed784f1d88d4d861fd90f9e8fc12901b9899f37b95bf55d","depends":[],"fee":50000,"sigops":8,"weight":1184},{"data":"01000000021d8ad160caffee6cb2330313221f6ae78ff771c5434bf13da523649749938184020000006b4830450221008ae0376a6614f336d427762a9ad6f3592ae67b9dda3cd501d3d0b8f04f3996c60220578cd084306a2afb29e4bfcfa04c089a7d5fe14c36ec4b7cbff406f0a20c7b2e0121020becb898f0c7f14395dd9ba6c55e54b0ee43fa4427ec704b43e5f970f7dce9b0ffffffff4f83c2ab08e6cf0b8950774eb79efe80b2d5d896f01dc86efe9dc6aa458b454d020000006a47304402200e5e440bfa0628214f7ae384028ea1883d14bff114aa1bb766590e154d285e1b02201eb9852f11dc4b35947cc7680b2f806f711ca9b24816e9c0f63be8f213028eca0121030a2ab6ecb3b36c4fb34081c4d93bfaa77c0b0ac1034814bf9773e1d4b8d2a59dffffffff03a0860100000000001976a91426d1cb933efb7d8ed02642e21f84adfdd33787c588acca340100000000001976a91426d1cb933efb7d8ed02642e21f84adfdd33787c588ac0000000000000000226a207b2274223a31322e36332c2270223a3936312e31382c2268223a32332e31337d00000000","txid":"8a9a41a1d40e4bfdcca2501b48cf3be93d7e9c1c9c1f204a06cc478ffb8cc9a6","hash":"8a9a41a1d40e4bfdcca2501b48cf3be93d7e9c1c9c1f204a06cc478ffb8cc9a6","depends":[],"fee":20950,"sigops":8,"weight":1664},{"data":"01000000018e23ccfb2aaae192acd7bf0674b63def37836925cd1d4d55bed632e0c4b985a2000000006b483045022100dd6c30f142d1ad4bc2038f9a88f4a856650d3f6c03423df66361457758429c79022061c61da427e3b3291aa8467ebb84ddc8444856140a79a5ae1ffd0ed5efbc88ec012103493bb2eab3c8b172e0b1ff3bd5320f7dac4d71ed398d04854fbfe1957b12ce82feffffff02203c0300000000001976a914e38027bb49c938beca19a3078caa67808d529a2e88ac7d091b00000000001976a9148eb9b1dcf7e2f4112502a4402c2e8b27f6b88acc88ac042e1000","txid":"e8cd45118d94081a91b6abf6f97fffe15920c22f1d92fc8641e36bbc3ceba79d","hash":"e8cd45118d94081a91b6abf6f97fffe15920c22f1d92fc8641e36bbc3ceba79d","depends":[],"fee":8718,"sigops":8,"weight":904},{"data":"01000000013013a87053728fe19219ca36256d908659938281adbb97440d7ca2f4b6749e4f000000008b48304502210090e3b2a91a33ab13c7d5a21fb28c4bdcdf48d752e1db4d057e7e8b73e53ad03702204ba02ac4333ea9ed957bc42c2092af173ada2b15d6c0d94d3f57347c1bfbe0270141048c9fca646144bc2dfca1769e47416e2a733f59a3dbce3ab3df7dc1e8ed86b135ea4782366b6ff67b73d6fdccef289b52d208eeb8c0ce41a7b5f3558be843d3a0ffffffff03ac221500000000001976a91418e45b290d8116093e8ce5c163bf53939c59449a88aceae43900000000001976a914bec582900041d0f4d2e432b3cb1dff9689067f1a88ac00000000000000001a6a185575372b5050626b7777612b626452725743567a41673d3d00000000","txid":"f3d6a221efcc1638a9df0370c173943890592ec5966558e719bd54c7364b9024","hash":"f3d6a221efcc1638a9df0370c173943890592ec5966558e719bd54c7364b9024","depends":[],"fee":10000,"sigops":8,"weight":1172},{"data":"010000000103d1113347dcb6ecc88bf03ae4cc07883ff8fa59c3fcf9f0b4155e35dfa3c4a000000000c1483045022100b5d76dcd841fd23aa87bb15ab4d1108ff7b5d925bad4a89b35ec4d34705d3c5502203c021dab3614ef39967ceab0c08f521406769826d68e73748ae73977ef18c825014c7674538763522102bc97dd13d20e7153fc9dc041d652e14b9b75a25ce40e886f276108d9e46ff27021038d7639c4163e71a8c3661c27b052051e4b21b33814166821af15edcd9700589b52ae6703042e10b1752103f4075272805bb4b6c19a03d08d0abf168dd9e34319f70b0d8348487bf37190f4ac680000000001c6c0f505000000001976a914d55220dafa3856d31d818c26a265010f12fd8ee888ac042e1000","txid":"4f5c56f1e8c2873018a67ce7606b266f9ec697e385042ee9df15c0eed1224a82","hash":"4f5c56f1e8c2873018a67ce7606b266f9ec697e385042ee9df15c0eed1224a82","depends":[],"fee":8250,"sigops":16,"weight":1112},{"data":"01000000015c622a405fa4efad94c02b8f77826a897a5074d3853dc9eacf2a1824248233f5010000006b483045022100b22f580d38617a66e4c615ed333700fec195d6c91b0e2a32781c7ffd7a81f21102201d651581e05f6a0117f092e7634df214f8655f2be92d858ce798d2b74082f393012103e817747947d076d94da1372859cd7ba00f96c667c8ff5c5a85ea1e3d720b0fe8ffffffff0178193101000000001976a914d7567817c2b85b27a0167d5676cdeab946c394bd88ac00000000","txid":"a7e9d65ff00ca078302142055800ca134fef8d43156e340c0330d854ff4fb560","hash":"a7e9d65ff00ca078302142055800ca134fef8d43156e340c0330d854ff4fb560","depends":[],"fee":5000,"sigops":4,"weight":768}],"coinbaseaux":{"flags":""},"coinbasevalue":156452918,"longpollid":"0000000000343071a5931b82c73966352a353bcac08c507da4567829fb4464a8451483","target":"00000000000009f98b0000000000000000000000000000000000000000000000","mintime":1482258198,"mutable":["time","transactions","prevblock"],"noncerange":"00000000ffffffff","sigoplimit":80000,"sizelimit":4000000,"weightlimit":4000000,"curtime":1482263912,"bits":"1a09f98b","height":1060357},"error":null,"id":0}

# request to "getwork"
> POST / HTTP/1.1
> Host: mc:18332
> Accept: */*
> Accept-Encoding: deflate, gzip
> Content-type: application/json
> X-Mining-Extensions: longpoll midstate rollntime submitold
> X-Mining-Hashrate: 2026000000
> Content-Length: 44
> User-Agent: bfgminer/5.4.2-17-g4f861b4

# response
> {"method": "getwork", "params": [], "id":0}
> HTTP/1.1 404 Not Found
> Content-Type: application/json
> Date: Di, 20 Dez 2016 19:58:32 GMT
> Content-Length: 76
> 
> {"result":null,"error":{"code":-32601,"message":"Method not found"},"id":0}

luke-jr, 2016-12-20:
It's libblkmaker you need to update for segwit, not BFGMiner.

hpbock, 2016-12-21:
Oops, thank you! Sorry for my noise.

closed 2016-12-21 by hpbock