luke-jr/bfgminer #687: r^pi3 error  when ./autogen.sh.... error: possibly undefined macro: AC_MSG_CHECKING

alberttwong, 2017-02-14:
raspberry pi 3 running on resin.io. 
```
root@raspberrypi3-ca56ba2:/usr/src/app/bfgminer# ./autogen.sh
Getting submodules...
Running autoreconf -if...
configure.ac:35: warning: macro 'AM_PATH_LIBGCRYPT' not found in library
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.ac and
libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:35: warning: macro 'AM_PATH_LIBGCRYPT' not found in library
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
configure.ac:17: installing './ar-lib'
configure.ac:16: installing './compile'
configure.ac:18: installing './config.guess'
configure.ac:18: installing './config.sub'
configure.ac:13: installing './install-sh'
configure.ac:13: installing './missing'
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.ac and
libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:32: error: possibly undefined macro: AC_MSG_CHECKING
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:34: error: possibly undefined macro: AC_TRY_LINK
configure.ac:39: error: possibly undefined macro: AC_MSG_RESULT
configure.ac:43: error: possibly undefined macro: AC_MSG_ERROR
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

alberttwong, 2017-02-14:
I bypassed this issue by just getting the binary. 
```
apt-get install -y bfgminer
```

marikan114, 2017-09-08:
Before ./autogen.sh make sure you have installed the following packages as well:

apt-get install -y pkg-config libgcrypt20-dev

Then it works!

Raginglikeaboss, 2017-09-18:
@marikan114 Thank you!  I ran into this problem for the first time last week and installing:
> libgcrypt20-dev

Fixed the issue for me immediately.

closed 2026-11-11 by alberttwong