luke-jr/bfgminer #692: Gekkoscience 2pac not working on bfg

j0nnay-aw83n, 2017-04-06:
Hey I just got my 2pac. And I can get it to work with cgminer but I want to use it on bfgminer is there anyway to do so?

luke-jr, 2017-04-06:
I haven't heard anything from them about this model.

j0nnay-aw83n, 2017-04-06:
Here's a link to the support for the device and they have a cgminer download as well https://bitcointalk.org/index.php?topic=1764803.0 
Just let me know if there's an easy work around

luke-jr, 2017-04-06:
I don't have one to work with, so you'll need to ask them. (I can only effectively support devices I'm sent a sample unit of.)

j0nnay-aw83n, 2017-04-06:
Okay, dam

j0nnay-aw83n, 2017-04-06:
Okay I've asked them. If they give me one I'll let you know just in case someone else wants/needs it

Raginglikeaboss, 2017-06-27:
@lukejr Have you received a Gekkoscience 2Pak USB miner from them yet?

If not I'd be willing to lend you one of mine if you're still willing to add support.

Please let me know!

luke-jr, 2017-06-27:
No, I didn't. But lending is too much trouble, and if it's not working, I may need specs etc which only they can provide anyway.

Raginglikeaboss, 2017-06-27:
@lukejr When I meant lending I simply meant I could throw in the extra postage and another shipping bag for you and you could hold onto it for as long as you needed.

I'd really like to see this merged and I've been in direct contact with the designer of these miners.  Maybe I could help by trying to get the information you'd need as well and hopefully put you two in direct contact?

luke-jr, 2017-06-27:
You could also just set me up with remote access to a Linux system with one plugged in... but temporary usage doesn't help with future problems/support others might have.

If the manufacturers can comment [here?] on what's changed since the original Gekkoscience device, that might be the most helpful next step.

MikeAZ65, 2017-11-08:
Hello, has there been any progress on getting the GekkoScience 2pac working in bfgminer 5.4.2 ?

Raginglikeaboss, 2017-11-09:
@luke-jr I can't say I'd ever let anyone remote connect into my computer, and as you said if anything changes wouldn't you then need to remote connect to my specially setup computer for you?

That sounds like much more trouble than me having had offered to send you one out of my own pocket and you can keep it for as long as you like.. 2 months, 2 years, however long you felt the desire to maintain GekkoScience USB miners in BFGMiner.

So instead, I'm offering to ship you one of the units I purchased for myself and you can keep it for all eternity.  All I would say is it would just be nice to receive a small tip in everyone's favorite cryptocurrency in return.

@sidehack had a single partner create a CGMiner version in under a week, but many of us prefer BFGMiner made by you - especially for Raspberry Pi mining and other form of mining using UNIX.

luke-jr, 2017-11-09:
> That sounds like much more trouble than me having had offered to send you one out of my own pocket and you can keep it for as long as you like.. 2 months, 2 years, however long you felt the desire to maintain GekkoScience USB miners in BFGMiner.

I mostly just don't care to deal with shipping it back. But if you put it that way, it makes sense to just call it a donation - I still support CPU/GPU/FPGA mining in BFGMiner, after all, so I doubt there's an end date to GekkoScience support either. :p

> So instead, I'm offering to ship you one of the units I purchased for myself and you can keep it for all eternity. All I would say is it would just be nice to receive a small tip in everyone's favorite cryptocurrency in return.

Okay. But just to set expectations: me tipping you for it would be essentially me paying you, for me to write code for you; I would think doing the coding work at no charge would be sufficient tip. (Perhaps other GekkoScience users who benefit from the code would be interested in pitching in via reimbursing you?)

luke-jr, 2017-11-09:
(If that's all fine with you, feel free to follow up via email <luke@dashjr.org> to get my shipping address.)

Raginglikeaboss, 2017-11-10:
I'll contact you via email this week regarding the shipping address and what days would be best to have it delivered/what carriers work best for you.

Honestly, I just wanted to say I received a tip from luke-jr.  If it was about money I'd more likely just sell it, I didn't mean to offend you or come off as begging.  I just thought it would be a fun thing to say that I helped continue BFGMiner's development and received a tiny tip or mention.

Afterall, while I'm giving away mining gear for free - it's you doing all the labor coding.  So nothing in return is expected.

Thanks for all your hard work!  I'll be in touch.

Snaps70, 2017-12-19:
Luke-Jr any news on what this will be supported? I am unable to get it to be recognized on any miner but Bitminter. BFG seems to detect it but fails to start it.

luke-jr, 2017-12-19:
Nothing ever came of the above offers. :/

sunk818, 2017-12-19:
Cgminer works fine for with 2pac except for zombie restarts. I could donate send you one of my 2pac but only if restarts can be supported by the driver or bfgminer. @sidehack @vthoang not sure if this is even possible. Scrpyt moonlander 2 recently added automatic restarts to non hashing devices.

luke-jr, 2017-12-19:
Without knowing what exactly goes wrong (at a technical level), I can't know if it's something software is capable of restarting.

jstefanop, 2017-12-19:
@sunk818 my board has a hardware level reset for the ASIC, that allows for the reset to work 99% of the time without needing full unplug. Trivial to implement, but depends on whether @sidehack has a reset connection through the UART. Pretty sure the chip he uses has a hardware reset as well.

Snaps70, 2017-12-19:
Sunk818- I can not get it to connect to cgminer either. I also get issues with ckpool test in it. So far Bitminer is the only one to recognize it and it comes up as a U3 miner. Then when I start mining I get Error codes for the stick over and over.

sunk818, 2017-12-20:
@snaps70 vthoang's cgminer works for me. He has binaries for windows. Linux and Mac is more involved. 

@luke-jr I'm willing to donate one for testing or development of you are willing to take a stab at it. It is a frustration that many of us using 2pac face every day with cgminer. My email is my GitHub user at Gmail if you give me a mailing address to send one to.

Snaps70, 2017-12-20:
sunk818 I have got it to boot finally in cgminer. still having pool failures......moving on to the next issue lol.

I will monitor as I prefer BFG.

Sethox, 2017-12-20:
I think the first guy that were willing to send it either has not send the gekkoscience 2pac, or he lives in another country so it will take another month to be delivered. Either way give the creator a server to work on (remote control) with the device plugged in and let the man access everything he needs to get some code work on. Just a suggestion based on what @luke-jr said.

sunk818, 2017-12-21:
I live in United States so like gets in few days. Whatever Luke wants. I don't care either way.

Sethox, 2017-12-21:
Well.. @sunk818 I suggest you contact @luke-jr directly and have a handshake (commit to a deal and confirm with each other) on it so it will happen. Because I don't think @luke-jr read EVERYTHING that happens on these issue posts.

sunk818, 2017-12-23:
@Sethox thx for your suggestion.

@jstefanop I am still willing to donate one, but confirmed with sidehack there is no hardware reset on the Gekko 2Pac. It seems you can't reset the hardware through the driver. I relegated myself to using a power time to turn the USB HUB on/off every few hours.

mikeharpe, 2018-01-04:
So is this going to get added or should I just switch to cgminer?

sunk818, 2018-01-05:
You can always donate your 2Pac if you want to get ball rolling... Would you like to do so?

mikeharpe, 2018-01-11:
I'm not going to just give you hardware. If you want to work on free software getting hardware is part of the cost of doing business.

sunk818, 2018-01-11:
> I'm not going to just give you hardware

As long as someone else is donating...

x-j0nnay-x, 2018-01-11:
@luke-jr I could set up a way for you to remote into my computer, but I would have to set up a new set up so it won’t be on my mining rig, it would be on a raspberry pi 3. Will that work for you?

mikeharpe, 2018-01-11:
I don't have an easy way to do that right now. I'd have to be home to set
that up. I'm in the US on Eastern time.

Can you maybe check back with me tomorrow? I'm off.

Thanks for checking back.

Michael

luke-jr, 2018-01-11:
> I'm not going to just give you hardware. If you want to work on free software getting hardware is part of the cost of doing business.

@mikeharpe Free software like BFGMiner is not a business. I donate my time to write and maintain it. If you want me to add support, the least you should be willing to do is donate hardware. The free software ideal would also have the person who wants the coding done, to do the code themselves and/or fund the development as well.

That being said, someone else has indeed donated one, so I will be looking into adding support soon.

sunk818, 2018-01-20:
Is there any way to get Gekko 2Pac to recognize with 5.5 with something else enabled in bfgminer? Even if the hash rate is slower it is okay. I just want to see if the USB is detected. This would be for Ubuntu Linux or Ubuntu (Windows 10 Linux Subsystem).

tchwpkgorg, 2018-02-28:
> That being said, someone else has indeed donated one, so I will be looking into adding support soon.

Any progress here? While it's generally agreed that Gekko 2Pac is no longer appropriate for mining, it's quite cheap hardware for learning purposes, generating blocks on testnet and such.

luke-jr, 2018-02-28:
Got tied up in some higher priority stuff, but should be getting back to 2Pac soon.

Sethox, 2018-02-28:
We know when this device is "done", when the owner closes this "issue". Then other issues may occur (as per usual).

joelhuebner, 2018-05-02:
Guys, with a USB 2.0 hub with plenty of power +90 WATTS.  the Gekko 2Pac is a pretty stable device.  I use Raspian-Streatch on a PiZero W to a Pi3b, it is just the gateway that connects your Gekko's to the pool you are using.   Yes you must use cgminer 4.1.0, available on a couple threads here.  this is not a plug & play 1 hour procedure!  you have to read the readme's, in there are the list of dependancies, these have to be manually added.  Then once you've gotten cgminer 4.1.0 from GitHub, you can extract, process & compile the code.  I've been messing with these for over 6 months, and I have one "burp" yet to smooth out.  I'll be happy to attempt to assist.  I DO NOT use Windows, I use Raspian, on Raspberry Pi's.   Good luck!  JLH

Alicolliar, 2018-10-09:
Have there been any further advancements on this?
Just got one of these and I was just wondering

sunk818, 2018-10-09:
@luke-jr I can send you a good working unit if you need another test unit. I think the one I sent you worked, but maybe the LED was not working right.

emx0r, 2018-11-16:
Hi, I'm not sure if it helps someone but anyways....
I got BFGminer working with 2pac. All you need to do is to firstly start cgminer, then you can kill it and if you start BFGMiner it recognizes 2pac and works perfectly.
Reason I'm doing it this way is cgminer is crashing my raspberry pi and workaround has helped me. BFGMiner is stable

sovos15, 2019-06-28:
I have 2pac BM1384's and was wondering if they will ever be supported?

Raginglikeaboss, 2019-07-15:
I have a handful of these units that I'm not using at the moment.  I don't buy & sell hardware much anymore.

@luke-jr If you're up for adding support for them into bfgminer, **I'm happy to donate one**.  No need to ship it back.  If it helps the hobby mining community I'm all for it.

Let me know what's easiest for you.  I plan on simply shipping it via USPS - so once I get your information and drop it off it'll arrive within a couple of days.

rob-2g, 2019-12-18:
@luke-jr do you have an update on this? I would like to use BFGMiner with some solo miners (GekkoScience NEWPAC (Dual BM1387) USB Stickminer 90+GH/s) I have. What do you need to go forward?

joelhuebner, 2020-03-04:
"Use the force"   v. 4.10 had the correct drivers for the Gekko 2.  I had my  10 * 2=20 processors running for years.  That model is depreciated, (hopefully, the price has too) Using cgminer on a Rpi B+ (network IO is the pig) nothing much else.  Perfect for a B+

heisreald, 2023-02-12:
New interest... 8D
I'm running a GekkoScience Compac F BM1397 @ 550Mhz getting 400-500GH/s with https://github.com/ckolivas/cgminer
I host a bitcoind and  getting BFGMiner up for a solo pool. I want to get away from soloCK and would like to run USB devices through BFG, in addition to my Antminers.
Idk what I can contribute, but I'm willing