luke-jr/bfgminer #695: [QUESTION] Cross-building

Tutul-, 2017-04-30:
Hi there, I just wondering if it's possible to cross-compile it from Linux to run on Windows (10 x86_64) ?
I don't have enought space on my windows to setup the build environment.

I want to compile it to include CPU support, I know, it's not verry efficient but it's only for testing purpose :)

luke-jr, 2017-04-30:
Yes, it is possible. Official builds are made this way.

closed 2017-04-30 by luke-jr

sunk818, 2018-02-26:
Can you provide a resource for compiling bfgminer under Windows 10 64-bit using Windows Subsystem for Linux (WSL) -> Ubuntu 16.0.4 LTS ?

I tried using some instructions from here:
https://github.com/HTMLCOIN/HTMLCOIN/issues/15

bfgminer compiles okay and runs fine as a Linux binary, but I can't seem to create an EXE for Windows