luke-jr/bfgminer #698: issue with Nanofury NF1

jedimaster77, 2017-06-07:
I am a noob, 

added a Nanofury NF1 to bfgminer

NFY 0: bitfury_init_oldbuf: Giving up after 4 tries

this repeats forever I have to close it and unplug the nanofury, the LED on the nanofury is amber if that helps.

Raginglikeaboss, 2017-06-27:
Did you ever manage to get your miner running?

As I recall these miners require some odd little additional code in the batch file in order to run properly.

I'm on mobile at the moment, but if you still need help I'd be happy to link you to some guides I use for customers when I used to sell these miners.

jedimaster77, 2017-06-28:
I did thanks, turned out the one I had was bad.  the person I got it from couldn't even make it work so he replaced it.

thank you for responding

j0nnay-aw83n, 2017-06-28:
i found some times the drivers need to be reinstalled and it works, i just had 2 of mine shit out on me, then i uninstalled all 3 of them then i rebooted my pc then reinstalled them worked fine. drive might not be dead but might just need some tlc

Raginglikeaboss, 2017-06-28:
@jedimaster77 Awesome!  I know how frustrating certain USB miners can be so I'm always willing to help when I'm knowledgeable enough to do so.