luke-jr/bfgminer #699: BFGMiner Not Detecting Futurebit Moonlander

scottdmat, 2017-06-18:
Hello,

New to BFGMiner and just got a few Futurebit Moonlanders. I setup a little mining rig using a pluggable powered USB hub and a Intel NUC running Windows 10-1607 64bit.

I'm able to get the drivers installed and have them show up in device manager as COM3 and COM4:
[image]

However, when I run BFGMiner no matter what I put to try and add them(COM3, COM4, auto, all) they never get detected. The only thing that picks up is my CPU when using "all".

Any advice or guidance would be great. Thanks!

luke-jr, 2017-06-18:
I have never heard of "Futurebit Moonlanders", and they're therefore not supported...

jstefanop, 2017-06-18:
@scottdmat support for moonlanders was never meeged to the main bfgminer branch. Check out my fork if you want the source, or download the precompiled binaries on the bitcoin/litecointalk threads.

luke-jr, 2017-06-18:
@jstefanop Why not merge it in?

jstefanop, 2017-06-18:
@luke-jr IIRC I had to do some "un-neat" changes to the low level com driver to be make it compatible with custom hardware I had on the board.

luke-jr, 2017-06-18:
Maybe we can work on fixing that up in the process?

metehaansever, 2017-06-18:
Hey and how can i help you

Raginglikeaboss, 2017-06-27:
Would love to see this merged.  These FutureBit MoonLanders are very well made and fun Scrypt miners.

I let friends borrow these to learn more about mining & proof-of-work in general.  It would be great to have BFGMiner support then by default, without needing to send my friends a link to the fork.