luke-jr/bfgminer #706: bfgminer does not filter-out bad incoming pool data, allows gridseed miners to lock-up.

VanessaE, 2017-07-07:
When a pool is having problems, sometimes it'll feed gibberish to the miners, or at least they end up with data they can't process.  I guess bfgminer does not attempt to "filter out" useless share data?

In any case, when that happens, bfgminer declares a unit as sick, then eventually dead.  Typically, that'll be all of my devices all at once.  I can't blame that on hardware. :stuck_out_tongue:

The only way out in this situation is usually to power-cycle all of the stuck devices.

VanessaE, 2017-07-08:
I've got a debug log containing one such event, but I can't really make sense of it (that is to say, nothing looks especially egregious).  It's a pretty big file (about 22 MB raw text), so if you want an excerpt from it, I'll be glad to oblige.  This log does not have "device protocol dump" turned on.

luke-jr, 2017-07-08:
Why is your pool sending you bad data in the first place?

I don't think @nwoolls (scrypt maintainer) is still active... :/

VanessaE, 2017-07-08:
That I am not sure of - the pool is Prohashing.com (stratum+tcp://prohashing.com:3333#skipcbcheck) and they have been having capacity and abuse problems lately, so I'm guessing something just plain breaks from time to time, sending bad data to the miners.

It doesn't happen super often; I might get a few such glitches in one day, or it might go several days between events.

mutluit, 2017-07-09:
With some/most drivers the .reinit_device function pointer is not set, ie. there is no code present for reinitializing the device.
IMO the SICK condition is mostly a false-positive, so if the above func would be present then auto-switching from SICK to the WELL condition will work. 
I did it experimentally for the icarus driver and it worked, but I'm currently no more working with the icarus driver as I switched to a different device and have some other problems with it... :-(

VanessaE, 2017-07-10:
Sometimes if bfgminer declares a device as "sick" it will eventually become well again later, but if it remains that way for too long it will become locked out, even if bfgminer doesn't actually declare it as "dead".  At that point, restarting bfgminer does not work (the device isn't recognized at all after that), and I have no choice but to power-cycle the device.

VanessaE, 2017-07-12:
It's not the pool, it appears to be hardware related on my end.  Bad USB hub, by the looks of it.

closed 2017-07-12 by VanessaE