luke-jr/bfgminer #710: bfgminer produces very low hashrate with AntMiner S3+ (openwrt ar71xx)

mutluit, 2017-07-22:
With this start script:
start_bfg.sh:
PAR="-o http://XXX:8332 --coinbase-sig 'XXX' --generate-to XXX --userpass user:XXX \
     --no-local-bitcoin --no-default-config --debuglog --no-longpoll --no-stratum --queue 8192 --scan-time 9999 --expiry 9999 --expiry-lp 9999 \
     --per-device-stats --show-processors \
     --scan bitmain:/dev/bitmain0 --set bitmain:model=S3 --set bitmain:layout=2:16 --set bitmain:timeout=18  --set bitmain:clock=225 --set bitmain:reg_data=0882 \
     "
TZ=UTC ./bfgminer $PAR 

Ie. testing solo mining to local bitcoind.
bfgminer reports such a low hashrate:

 bfgminer version 5.4.2-7-g9f781b8 - Started: [2017-07-22 17:05:25] - [  0 days 00:03:20]
 [M]anage devices [P]ool management [S]ettings [D]isplay options                                                                                                                                 [H]elp [Q]uit 
 Pool 0: XXX         Diff:805G  - GBT   LU:[17:05:24]  User:XXX
 Block #477036: ...f6828673  Diff:805G ( 5.76E)  Started: [17:05:24]  I: 0.00 BTC/hr
 ST:42  F:0  NB:1  AS:0  BW:[ 18/  0kB/s]  E:0.00  BS:123k
 1            | 195.7/199.9/199.9Gh/s | A:0 R:0+0(none) HW:0/none
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 BTM 0:       | 196.3/199.8/200.0Gh/s | A:0 R:0+0(none) HW:0/none
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 [2017-07-22 17:05:22] BitMain Detected: /dev/bitmain0 (chain_num=2 asic_num=16 timeout=18 freq=225-225 volt=0725-0725)
 [2017-07-22 17:05:22] Probing for an alive pool
 [2017-07-22 17:05:22] Testing pool http://37.139.71.2:8332
 [2017-07-22 17:05:24] Network difficulty changed to 805G ( 5.76E)
 [2017-07-22 17:05:24] New block: ...3eb3a0d2738682f6 diff 805G ( 5.76E)
 [2017-07-22 17:05:24] Selected getblocktemplate protocol for pool 0
 [2017-07-22 17:05:24] Pool 0 http://37.139.71.2:8332 alive
 [2017-07-22 17:05:25] BitMain: Opened on /dev/bitmain0
 [2017-07-22 17:05:25] Block height change to 477035 detected on pool 0

Normally, with default device settings (ie. no overclocking) it should be more than 450 GH/s.
What's wrong?

sunk818, 2018-01-29:
Have you tried a different miner like cgminer and gotten higher hash rate? What happens if you remove all the parameters and just use the required ones?

jiluyaku, 2018-04-20:
Hello, @mutluit 
May I seek your help on how to run a BFGMINER on Antminer? I am figuring out how to run the bfgminer on my antminer s5, but I haven't got any idea to start with at the moment. Thank you!