luke-jr/bfgminer #715: bfgminer-5.4.2: no 'configure' file

dchmelik, 2017-09-17:
The [bfgminer-5.4.2.tar.gz](https://github.com/luke-jr/bfgminer/archive/bfgminer-5.4.2.tar.gz) file (which gets saved as bfgminer-bfgminer-5.4.2.tar.gz, which you could have checked in relation to issue #714, has no 'configure' file. Please add some correct documentation (preferably in each source code archive file, maybe above usage instructions, if the archive file doesn't have a binary executable file anyway) or the files current documentation refers to (configure.)

luke-jr, 2017-09-17:
Spamming will not help you. The documentation is correct. Your link is wrong.

closed 2017-09-17 by luke-jr

dchmelik, 2017-09-17:
So, this isn't your project https://github.com/luke-jr/bfgminer/archive/bfgminer-5.4.2.tar.gz ?  That's what my link was.  It appears either you didn't see that's a sub-directory of this project (that I'm posting this as an issue on) or you're the spammer/troll.