luke-jr/bfgminer #721: 100% CPU usage if a device goes offline/disconnects

VanessaE, 2017-10-03:
This is bfgminer compiled from source, at commit 9f781b872db9a91f741bb8d08d1e60383a84a0b2, running on Debian 9.1, controlling several gridseed "orbs" and blades.  CPU and GPU mining are not enabled (at least not intentionally).

If one of my devices disappears, for example if its USB cable gets unplugged (or stepped on by my cat :stuck_out_tongue_closed_eyes: ), the kernel recognizes the disconnect event and reports it via `dmesg`, but this causes bfgminer to immediately start spinning at least one CPU core at 100%.

If I were to then simply shut off the USB hub, or unplug its main USB cable from the computer, that of course would cause several devices to disappear all at once.  bfgminer seems to hog one CPU core per device that goes away, so it'll spin all six of my cores at 100% on such an event.

bfgminer does indicate that the disconnected devices are failing to return work, but it does not mark them as "sick" or "dead" or otherwise gracefully disconnect from them.

Since bfgminer is trying to keep its in-use device names "open", reconnecting a missing device causes the kernel/udev/whatever to give it a new name.  bfgminer won't be looking for that new name (it'll usually be out of range as implied by the script below), so it doesn't react to the reconnect event.

bfgminer will continue to eat CPU indefinitely, until I exit.  I can simply press 'q' or Ctrl-C when it gets into this state, so it's not hard-locked.  My keepalive script restarts bfgminer after a couple of seconds, though I usually have to kill the script, reset my devices, and restart it to get them all numbered right so bfgminer will see them.

```bash
#!/bin/bash

cd /home/vanessa/coin-mining-stuff/bfgminer-official

while true; do
	timed-process 86400 \
		./bfgminer \
			--debuglog \
			--scan-time 5 \
			-S gridseed@/dev/gridseed0 \
			-S gridseed@/dev/gridseed1 \
			-S gridseed@/dev/gridseed2 \
			-S gridseed@/dev/gridseed3 \
			-S gridseed@/dev/gridseed4 \
			-S gridseed@/dev/gridseed5 \
			-S gridseed@/dev/gridseed6 \
			-S gridseed@/dev/gridseed7 \
			-S gridseed@/dev/gridseed8 \
			-S gridseed@/dev/gridseed9 \
			-S gridseed@/dev/gridseed10 \
			-S gridseed@/dev/gridseed11 \
			-S gridseed@/dev/gridseed12 \
			--scrypt --failover-only \
			--set gridseed@6D8038644857:clock=875,chips=5 \
			--set gridseed@6D9634674857:clock=875,chips=5 \
			--set gridseed@6D9526874853:clock=900,chips=5 \
			--set gridseed@6D8F489B5748:clock=863,chips=5 \
			--set gridseed@6D7320865251:clock=863,chips=5 \
			--set gridseed@6D7835994857:clock=863,chips=5 \
			--set gridseed@48F66F773137:clock=800,chips=40 \
			--set gridseed@490153513135:clock=825,chips=40 \
			--set gridseed@48EB82553734:clock=825,chips=40 \
			--set gridseed@48F086473037:clock=825,chips=40 \
			--set gridseed@48E4556C3030:clock=825,chips=40 \
			--set gridseed@48FD746F3137:clock=825,chips=40 \
			<username and password arguments omitted>
	wait
	echo -e "\n\nKeepalive script detected exit, restarting...\n\n"
	sleep 2
done
```

I only explicitly enabled scrypt and gridseed drivers at compile time, but of course a bunch of other stuff got compiled in by default, so here is config.h:

```h
/* config.h.  Generated from config.h.in by configure.  */
/* config.h.in.  Generated from configure.ac by autoheader.  */

/* List of drivers being built */
#define BFG_ALGOLIST " SHA256d scrypt "

/* List of drivers being built */
#define BFG_DRIVERLIST " antminer avalon avalonmm bfx bifury bigpic bitforce bitforce:pci bitfury cairnsmore cointerra compac drillbit dualminer erupter gridseed hashbusterusb hashfast icarus klondike littlefury modminer proxy proxy:getwork proxy:stratum rockminer twinfury x6500 zeusminer ztex "

/* List of drivers being built */
#define BFG_LOWLLIST " ftdi pci pci:uio pci:vfio spi usb vcom "

[... 694 lines cut ...]
#else
# define _GL_UNUSED
#endif
/* The name _UNUSED_PARAMETER_ is an earlier spelling, although the name
   is a misnomer outside of parameter lists.  */
#define _UNUSED_PARAMETER_ _GL_UNUSED

/* The __pure__ attribute was added in gcc 2.96.  */
#if __GNUC__ > 2 || (__GNUC__ == 2 && __GNUC_MINOR__ >= 96)
# define _GL_ATTRIBUTE_PURE __attribute__ ((__pure__))
#else
# define _GL_ATTRIBUTE_PURE /* empty */
#endif

/* The __const__ attribute was added in gcc 2.95.  */
#if __GNUC__ > 2 || (__GNUC__ == 2 && __GNUC_MINOR__ >= 95)
# define _GL_ATTRIBUTE_CONST __attribute__ ((__const__))
#else
# define _GL_ATTRIBUTE_CONST /* empty */
#endif

#ifdef WIN32
#include "winhacks.h"
#endif
```

luke-jr, 2017-10-03:
Sounds like an issue with the driver. Since the scrypt maintained disappeared, you may need to submit a patch to fix it to make any progress on it.

jstefanop, 2017-10-24:
I can confirm this issue exists on my drivers as well...I don't think its driver related, but ungraceful exit of vcom driver/way bfgminer deals with USB disconnects. Ill dig deeper.

VanessaE, 2017-10-31:
In the meantime, @luke-jr can you at least fix the problem of bfgminer not honoring patterns in device strings?  One ought to be able to do something like `-S gridseed@/dev/gridseed*`, but nothing I've tried in that regard works (see https://github.com/luke-jr/bfgminer/issues/610#issuecomment-301312616 for details).  That would at least work around the problem of requiring one to cycle all of the miners' USB connections to bring back ones that briefly drops out.

VanessaE, 2017-12-13:
Can something please be done about these issues?  This 100% CPU problem is quite old and very infuriating.

Considering there have been no commits for well over a year, should bfgminer be considered an abandoned project?

VanessaE, 2019-07-13:
Guess this has been abandoned (and gridseeds are unprofitable most of the year now, anyways)

closed 2019-07-13 by VanessaE