luke-jr/bfgminer #725: Installing BFGMiner on Raspberry Pi 3 B

tartoga1, 2017-10-24:
Things were going well until the last 2 steps before start mining. Did the clone ok. Did the cd bfgminer and entered the next step ./autogen.sh. It started doing things and I got the following:
running autoreconf -if ...
./autogen.sh:24 ./autogen.sh: autorefig: not found

I'm a newbe just experimenting with this before I start spending a lot more of my grand children's xmas money ;-)
I have installed successfully minera, but this one and minepeon are giving me trouble ;-(

tartoga1, 2017-10-24:
Well, I'm getting closer. I used an image to install the OS on the raspberry. The install went a lot further. The following are the issues now;
Updating version.h...
cmp: version.h: No such file or directory

checking for libmicrohttpd... no

configure: WARNING: libmicrohttpd 0.9.5+ not found; getwork proxy will be unavailable
checking for libevent... no
configure: error: Package requirements (libevent >= 2.0.3) were not met:
No package 'libevent' found

I have attached a copy of the results hoping someone may point me in the correct direction ;-)
[Installing BFGMiner results.txt](https://github.com/luke-jr/bfgminer/files/1409790/Installing.BFGMiner.results.txt)

tartoga1, 2017-10-24:
anyone out there seeing this post...

tartoga1, 2017-10-24:
I closed by accident ;-(

jstefanop, 2017-10-24:
You don't have the correct dependancies installed...go over the readme and make sure you install all required libraries before compiling.

closed 2017-10-24 by jstefanop

tartoga1, 2017-10-24:
Thanks for the reply. I’m not sure what readme you are referring to. I followed the instructions step by step. The best results I’ve got was loading an image rather then the noobs install and got almost done. I don’t see a readme file ;-(

 

David

 

Closed #725 <https://github.com/luke-jr/bfgminer/issues/725> .

tartoga1, 2017-10-24:
I have a lot of notes and have gone through them again. Other then following the instruction from the web, I can't find a readme. I thought the commands whould load whatever is required. Except of course when near the end it failed.

David

tartoga1, 2017-10-24:
HI, good thing I have a number of class 10 8gb microsd’s ;-)

Minera loads successfully.

 

Should I try copy’g the bfgminer 5.4.2 that’s on your readme area?

 

David

tartoga1, 2017-10-25:
Getting closer.

 

Doing the ./configure

Makefile.am:40: warning: ‘INCLUDES’ is the old name for ‘AM_CAPPFLAGS’ (or ‘*_CPPFLAGS’)

Makefile.am: installing ‘./depcomp’

Parallel-tests: installing ‘./test-driver’

Updating version.h…

Cmp: version.h: No such file or directory

 

Maybe I pay you to get me past the hiccup ;-)

 

David