luke-jr/bfgminer #729: Openwrt Support for MVEBU Platform

Grntrenchman, 2017-11-19:
As [per instructions](https://github.com/luke-jr/bfgminer/blob/bfgminer/README.OpenWrt), raising a issue for a unsupported platform.

I run a Linksys WRT1900AC, with a Marvell SOC. Let me know if there is any information I can provide to help.
I have block buster asics, and have installed the correct driver. The asics show in /dev/ as ttyUSB.

luke-jr, 2017-11-19:
What version of Openwrt are you using?

Grntrenchman, 2017-11-19:
Chaos calmer, 15.05.1 r48532

luke-jr, 2017-11-22:
Please try the new packages and confirm it works for you.

Grntrenchman, 2017-11-22:
It works! Installed and running behind screen. At first I couldn't install
it due to a lack of signature file, but otherwise runs fine.

opkg denied the entry once it 404'd on Packages.sig
I disabled the signature check temporairly to install, but you can generate
files yourself <https://forum.openwrt.org/viewtopic.php?pid=297427#p297427>:

In order to get the verification work for self-made packages, following
below steps to generate key files and sig file:

Generate a private and public key files:
usign -G -s mime.key -p mime.pub
Add mime.pub file into opkg’s keys:
opkg-key add mime.pub
Using private key file to generate signature file for Packages
usign -S -m Packages -s mime.key -x Packages.sig
Put Packages and Packages.sig into the repository packages folder
Run opkg update to versify whether it accepts your feed