luke-jr/bfgminer #730: --shares <arg>      Quit after mining 2^32 * N hashes worth of shares (default: unlimited)

sunk818, 2017-12-12:
Can someone explain how `--shares` works?

I cgminer 4.10, it was a simple value of accepted hashes. Each accept hash is the same value as the current pool difficulty. If you submit an acceptable share with 256 pool difficulty, your shares go up by 256.

I'm using bfgminer 5.4.2-27 (jstefanop Moonlander 2 specific) and `--shares` does not make sense to me.

2^23 is 8388608. If I multiply that by the number of hashes worth of shares (whatever that means).

The whole thing just stinks of confusion.

Is the way scrypt accepted hashes work different from sha256d?

luke-jr, 2017-12-13:
> I cgminer 4.10, it was a simple value of accepted hashes. Each accept hash is the same value as the current pool difficulty. If you submit an acceptable share with 256 pool difficulty, your shares go up by 256.

Sounds about the same...

> Is the way scrypt accepted hashes work different from sha256d?

No, but scrypt is a very slow algorithm, so typically you would probably be specifying some number like 0.0001 or something.

jstefanop, 2017-12-13:
bfgminer internally treats a scrypt diff 1 shares as 1/65526th of a SHA256 share. This is why you see "Diff 134m/7m" which is a milli share.

sunk818, 2017-12-13:
Yeah the whole concept of accepted shares is very confusing. On multi pool, they set the share difficulty at 35m. If I send an accepted hash, the counter goes up by 1 instead of 35. So, I'm trying to restart bfgminer after x number of shares to give the miners a cool down period.

My expectation is that accepted shares increment by current pool difficulty x accepted share. Pools payout by the same method. So, having bfgminer not adhere to how most (all?) pools payout just adds to the confusion.

Is it me, or doesn't that seen most logical to display like how cgminer 4.10 does?

Forgive me if I don't understand how sha256 and scrypt algorithm works and the answer is obvious due to the difference.

sunk818, 2017-12-13:
I see 65536 or 2^16. We still don't easily know what we've accumulated in terms of accepted shares. It just shows the count of shares accepted not sum)share difficulty x accepted share).

Was trying to give the USB asic a cool down after so many shares but was confused.

So if an accepted share is 7m, what is the numeric value for that?
If I submit 10 shares a minute at 7m, what would the share value  (--shares) be at the 30 minute point?

luke-jr, 2017-12-13:
https://en.wikipedia.org/wiki/Metric_prefix

> So if an accepted share is 7m, what is the numeric value for that?

0.007

> If I submit 10 shares a minute at 7m, what would the share value (--shares) be at the 30 minute point?

2.1

sunk818, 2017-12-13:
Okay, thank you. That helps me a lot. I'll try to configure with this clarification. Thanks very much.

closed 2017-12-13 by sunk818