luke-jr/bfgminer #732: Error at 'git clone'

SNoyes32, 2017-12-17:
When attempting follow steps for the initial build of an ASIC miner attached to a Raspberry Pi, the line:
git close https://github.com/luke-jr/bfgminer.git gives the following error:
In[12]:= git clone https://github.com/luke-jr/bfgminer.git                      

Syntax::sntxf: "git clone https:" cannot be followed by 
    "//github.com/luke-jr/bfgminer.git".
Also tried git clone https://github.com/luke-jr/bfgminer with the same (well similar since the .git is missing) results.

Any help would be greatly appreciated, I grew up in a windows world, but I don't want to build GPU miners if I can avoid it!
Thanks,
-Shawn

sunk818, 2017-12-17:
Try git clone

Add in copy not close. And you don't need the .git at end

sunk818, 2017-12-17:
Try git clone

Add in copy not close. And you don't need the .git at end

If you want to build for moonlander 2 asic jstefanop has his own fork

luke-jr, 2017-12-18:
Sounds like you're using some math program, not a shell...

closed 2017-12-18 by luke-jr